eelcocramer/node-bluetooth-serial-port #137: Getting 'cannot connect' and 'connected' at the same time

ultamatt, 2017-01-24:
Hey guys! Trying to do a simple bluetooth connection example here and getting the following console log out. (along with xcode bluetooth tools.)

[image]

Here's my server code: 

```
router.get('/find', function(req, res, next) {
    var btSerial = new(require('bluetooth-serial-port')).BluetoothSerialPort();
    console.log("finding");
    btSerial.on('found', function(address, name) {
        console.log("found " + address + " - " + name);
        btSerial.findSerialPortChannel(address, function(channel) {
            console.log("found channel " + channel);
            console.log("trying to connect");
            btSerial.connect(address, channel, function() {
                console.log("connected to " + address + " - " + channel);

                btSerial.write(new Buffer('example my data', 'utf-8'), function(err, bytesWritten) {
                    if (err) {
                        console.log("error" + err);
                    } else {
                        console.log("wrote" + bytesWritten);
                    }
                });

                btSerial.on('data', function(buffer) {
                    console.log("read" + buffer.toString('utf-8'));
                });
            }, function() {
                console.log('cannot connect');
            });

            // close the connection when you're ready
            btSerial.close();
        }, function() {
            console.log('found nothing');
        });
    });

    btSerial.inquire();
    res.render('index', {
        title: 'Finding'
    });
})
```

So... What gives? how come my mac decides that it's connected but the library says that it can't?

ultamatt, 2017-01-24:
```
bash-3.2$ npm -v
3.10.3
bash-3.2$ node -v
v6.3.1
```

eelcocramer, 2017-01-25:
It is possible that the Bluetooth radios of both devices are connected but the services (the client and server software) are not connected.

closed 2017-03-20 by eelcocramer