eelcocramer/node-bluetooth-serial-port #141: Feature request: Stop scanning and callback at first found device!

sandangel, 2017-03-02:
I have to wait until inquire execution finished and when I repeating scanning until found, I can't stop sending request.

eelcocramer, 2017-03-03:
Most bluetooth SDKs won't let you start and stop the inquiry process. On Linux it is possible to specify a search time that is currently hardcoded around 10 seconds (1.28 * 8). It is also possible to specify how many devices should be search for at max. The inquire function might return before the timer has ended if the maximum number of devices has been found but I did not test that.

On Linux it is not possible to emit results before this function returns to the results will be emitted almost at the same time.

On OSX it might be possible to emit a result the moment is has been found.

sandangel, 2017-03-04:
[https://github.com/sandeepmistry/noble](url)
 I have just migrated to bluetooth-serial-port lib from noble, a BLE library since I don't need to use BLE device anymore. As in usage instruction, I can use noble.startScanning and noble.stopScanning to manage scanning process. And in scanning process, .on('found') event was emitted as soon as first device was found. So I made these features request. Do you mean that start and stop inquiry process is not possible with SPP profile?

eelcocramer, 2017-03-04:
This has nothing to do with the SPP profile. It is possible to implement what you are requesting although on some platforms (OSX) it might be easier to implement than for others (Linux).

I can see why this would be useful and I would be happy to merge a PR that implements this.

sandangel, 2017-03-05:
OK, I 'll take a look on your code.

shuangjj, 2017-04-05:
It would be great to specify the inquiry timeout in the library.

closed 2017-09-23 by sandangel