eelcocramer/node-bluetooth-serial-port #144: Server: Listen in multiple channels simultaneously.

atilag, 2017-03-06:
It would be nice if we can handle multiple connections in the server code, so we can listen on different channels at the same time.

meisterlampe, 2017-05-05:
Any news on this?

atilag, 2017-05-15:
Nop, sorry. Running out time lately... :(

meisterlampe, 2017-07-13:
Sorry, don't want to jangle your nerves, but I'm really waiting for this feature O:-) 
If you still have no time right now this is also ok of course. :) I only want to kindly ask for the current status.
Really appreciate your voluntary work on this project 👍

meisterlampe, 2017-10-10:
Hi it's me.... again.... ;-) 
Is there any chance, that this will be implemented?

atilag, 2017-10-10:
I'm afraid that I won't be able to implement it anytime soon, sorry.

meisterlampe, 2017-10-10:
OK, no problem. :)
Thank you for the fast answer and of course for the great work on this project!