eelcocramer/node-bluetooth-serial-port #158: Windows listPairedDevices Showing Unpaired Devices

obermillerk, 2017-07-12:
When calling listPairedDevices, there are a couple of cases where unpaired devices will show up in the results.  I'm running in Windows 10 Pro version 1703, using node 6.10.3.

1: After unpairing from a device, that device will remain in the results until the machine is restarted.

2: After opening the dialog to add a new Bluetooth device from the OS's Bluetooth settings, all devices found by the OS's device scan in this dialog show up in the results until the machine is restarted.  This does not require any action to be taken in the dialog, simply opening it and closing it after devices are found produces this behavior.

Here is a test script that exhibits this behavior: [bt-test.zip](https://github.com/eelcocramer/node-bluetooth-serial-port/files/1142573/bt-test.zip)

Case 1: When run on a fresh boot with a device paired, the output is the following, as expected:

    Searching for devices...
    Search finished.
    Found devices:
        RNBT-87BF  //  00:06:66:61:87:BF

After unpairing the device, the exact same output is produced.  Only unpairing does not add any other sensors than the one being unpaired.

Case 2: When run on a fresh boot with no devices paired, without touching Bluetooth settings, the output is the following, as expected:

    Searching for devices...
    Search finished.
    No devices found!

After opening the add Bluetooth device dialog, allowing it to find a few devices, and closing it without doing anything:

    Searching for devices...
    Search finished.
    Found devices:
        unknkown  //  D0:03:4B:1E:F4:C8
        RNBT-87BF  //  00:06:66:61:87:BF
        RNBT-88EF  //  00:06:66:86:88:EF

None of these are paired.

In the version of Windows that I'm running, the OS does not try to discover new Bluetooth devices on the same screen as the unpair functionality.  I also have tested this on a machine that is running Windows 10 Home version 1607 which searches for devices on the same screen as unpairing, so only case 2 behavior can be replicated.

I have not tested this on anything but these two Windows machines, so I can't say for sure if this issue is isolated to Windows.

barchard, 2017-07-12:
On macOS 10.12.5, this lists all paired devices (although none are actively connected):

```
node ./bt-test.js

Searching for devices...
Search finished.
Found devices:
    R-8317 // a0-b1-c2-d3-e4-f5
    MCR-B142 Yamaha // a0-b1-c2-d3-e4-f5
    Pebble Time C34D // a0-b1-c2-d3-e4-f5
    FireFly-8D45 // a0-b1-c2-d3-e4-f5
    Mouse // a0-b1-c2-d3-e4-f5
    R-83C1 // a0-b1-c2-d3-e4-f5
```

eelcocramer, 2017-07-13:
@checksummaster any idea?

eelcocramer, 2017-08-08:
I might have found a solution to this problem but as I do not have a windows environment available I would like someone to check my fix.

I'm filtering out devices by testing if the `BTHNS_RESULT_DEVICE_AUTHENTICATED` flag is set on result set containing the devices that were found by the OS.

```c
querySet->dwOutputFlags & BTHNS_RESULT_DEVICE_AUTHENTICATED
```

See [docs](https://msdn.microsoft.com/en-us/library/windows/desktop/aa362918(v=vs.85).aspx) for more details...

I've added this line to a branch called `issue-158` which I've just pushed to the repo. If you could test this to see if this works I would appreciate it.

obermillerk, 2017-08-08:
I'll try this when I can, might be a day or two.  Thanks!

eelcocramer, 2017-08-09:
👍🏼

obermillerk, 2017-08-14:
This worked like a charm! Unpaired devices no longer show up, and unpairing a device removes it from visible items.

eelcocramer, 2017-08-15:
Keeping this open untill it is merged and released.

eelcocramer, 2017-08-15:
Thank for the feedback!

obermillerk, 2017-09-28:
Hey, just wondering when this is gonna be published?

eelcocramer, 2017-09-28:
Sorry I forgot about this one. I was on vacation when you confirmed the fix. I will publish this soon!

obermillerk, 2017-09-28:
No worries, thanks!

eelcocramer, 2017-09-29:
Just released this as part of version 2.1.5

closed 2017-09-29 by eelcocramer