eelcocramer/node-bluetooth-serial-port #162: Close event does not actually represent disconnect

obermillerk, 2017-07-31:
Hi again,
I've encountered a new problem in my app.  My app needs to automatically close serial port devices, so we are depending on the 'closed' event to tell us that the disconnect actually occurred.  However, if I disconnect and then quickly attempt to reconnect the device stays connected in the OS but serial-port can't connect to it because it is already connected to "another" device, even though that device is the one I'm using.  Is it possible to wait for some kind of signal from the OS that the device actually is disconnected before emitting 'closed'?  As is, it appears that your code is very optimistic in this sense, as it just assumes that closing the connection succeeds and emits a 'closed' event (see [here](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/lib/bluetooth-serial-port.js#L125)).

eelcocramer, 2017-07-31:
Can you provide example code that can be used to reproduce this behaviour?

Does this happen on all platforms (darwin, windows & linux)?

obermillerk, 2017-07-31:
I'm not sure that I can provide code that reproduces this since it is dependent on trying to connect to a device. I'll try to put something together.

I assume that it will happen on all platforms since I think I can see the issue in the code ([here](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/lib/bluetooth-serial-port.js#L125)), however I'm only running Windows so I can't test others.  I think instead of emitting a 'closed' event immediately this should be waiting for a signal that the device is disconnected and THEN emit a 'closed' event.  I'm not sure if this is even possible, since I'm not very familiar with the underlying specs, but I should think that some signal is sent out from the OS when a device disconnects.

eelcocramer, 2017-07-31:
Though you are right on the fact that the return value for closed is not checked on any platform implementation and this is something that could be improved I want to refer to #90 as this might help you.

What would, in your opinion, be the expected behaviour when a call to `close` fails?

eelcocramer, 2017-07-31:
Closing happens synchronously so the `emit` is only called after the native close function did finish. However the native close function might fail, currently the return code of the native close is ignored. This is something that can be improved but I'm not sure in what state the connection would be when a call to close did fail.

eelcocramer, 2017-07-31:
On Windows the `[close_socket](https://msdn.microsoft.com/en-us/library/windows/desktop/ms737582(v=vs.85).aspx)` is called. Depending on the error code an assumption of the state of the connection can be made.

obermillerk, 2017-07-31:
I see what you're saying, that is a bit of a dilemma.

In regards to the close_socket call, is there a similar mechanism in other OS's? Perhaps this is uniform behavior that can be plugged into to determine the state after a failed disconnect? It also looks to me like all of those error codes seem to say that the disconnect didn't happen successfully, so it would still be connected, but this is a purely surface level evaluation on my part.

obermillerk, 2017-07-31:
Alternatively, since the main problem I have with this bug is that I can't reconnect to the device without disconnecting it manually (turning it off or removing it in system settings) is there some way to check if the device is already connected when trying to connect and just use that connection?  Right now it just rejects the connection because the device is already connected, even though it's connected to my system.

eelcocramer, 2017-07-31:
Did you try *not* reusing the connection and creating a new object like mentioned in #90 ?

Also please not that there is a difference between the radio connection that the OS has with the device and the serial port channel that this module opens. You are closing the latter where the first is managed by the OS. It should be possible to create another serial port connection while the radio connection still stands.

obermillerk, 2017-07-31:
I just did and I still get the same issue.  It seems to be on the third time I try to connect.  As in, connect once normally, disconnect then reconnect quickly, disconnect and reconnect quickly but reconnect does not work this time.  I'm investigating whether this could be related to something else in my app, since there is a high spike in CPU usage once this starts showing up.

eelcocramer, 2017-08-01:
Ok, in the meanwhile you could check if closing the connection fails or not. I've created a branch called `issue-162` which should print some messages to the console. You should see 2 extra messages:

1. That the close call was either successful or a failure. The latter will also print the status code
2. A message that the `close` event will be emitted.

They should appear in that order.

obermillerk, 2017-08-01:
So I've run this with your new branch and the disconnect is successful even though the issue still appears.  I'm thinking that this might be an issue with the device I'm connecting to since it is a custom device.  When connecting I have it use your findSerialPortChannel to make sure that we have the right channel, and I did some testing and this seems to be where the falloff happens, so I'm thinking it might be that the serial service is crashing on the device.  We're going to look into it from this angle and I'll get back to you. Thanks for your help so far though!

eelcocramer, 2017-08-02:
No problem & good luck

wayofthefuture, 2017-08-03:
I'm having the exact same problem over here. I'm going to do some reading on #90 and try to help with this issue if possible.

obermillerk, 2017-08-03:
So we tested this against some new hardware (the device we're connecting to that is) that we're going to be moving to, and it seemed to alleviate this issue on first testing.  I can't really test this very extensively though cause we don't have full units yet, just test boards.  I still maintain that the closed event should only be emitted once the device is actually disconnected, though I understand the state problem you mentioned.

obermillerk, 2017-08-03:
Also the high CPU usage I mentioned was something unrelated to this library, it seems.

wayofthefuture, 2017-08-03:
I read #90 and did some review and testing.

We are using a new instance of the serial port object for everything: including the inquire, find channel, and connect. On subsequent connects, it also uses new instances. All new instances.

Windows 10: the device does disconnect and we can reconnect immediately.
Ubuntu Linux 17.04: the device does not disconnect (sometimes) and we cannot reconnect immediately.

From what I can remember, calling device.close() will remove any "close" events on the object. So if you add a close event, then call device.close(), the event won't get fired (I think).

Just in case you were wondering the available inherited properties you can use this: for (let p in device) console.log(p);

Device Object Properties:
[SERVICES] 12:01:41 domain
[SERVICES] 12:01:41 _events
[SERVICES] 12:01:41 _eventsCount
[SERVICES] 12:01:41 _maxListeners
[SERVICES] 12:01:41 inq
[SERVICES] 12:01:41 found
[SERVICES] 12:01:41 finish
[SERVICES] 12:01:41 address
[SERVICES] 12:01:41 buffer
[SERVICES] 12:01:41 connection
[SERVICES] 12:01:41 isReading
[SERVICES] 12:01:41 listPairedDevices
[SERVICES] 12:01:41 inquire
[SERVICES] 12:01:41 inquireSync
[SERVICES] 12:01:41 findSerialPortChannel
[SERVICES] 12:01:41 connect
[SERVICES] 12:01:41 write
[SERVICES] 12:01:41 close
[SERVICES] 12:01:41 isOpen
[SERVICES] 12:01:41 setMaxListeners
[SERVICES] 12:01:41 getMaxListeners
[SERVICES] 12:01:41 emit
[SERVICES] 12:01:41 addListener
[SERVICES] 12:01:41 on
[SERVICES] 12:01:41 prependListener
[SERVICES] 12:01:41 once
[SERVICES] 12:01:41 prependOnceListener
[SERVICES] 12:01:41 removeListener
[SERVICES] 12:01:41 removeAllListeners
[SERVICES] 12:01:41 listeners
[SERVICES] 12:01:41 listenerCount
[SERVICES] 12:01:41 eventNames

eelcocramer, 2017-08-03:
I've added a commit to the `issue-162` branch that calls a function called [`shutdown`](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740481(v=vs.85).aspx) which should disable the socket. Not sure if this is the desired behaviour but IMO worth checking it out.

On OSX there is something similar, this is currently already being called. 

On Linux I could not find anything yet.

Also: does the remote side (server) also close the server socket when the remote client disconnects?

wayofthefuture, 2017-08-04:
I wasn't experiencing the issue with Windows, only Linux. Could you clarify your last question?

eelcocramer, 2017-08-04:
See [this code example](https://people.csail.mit.edu/albert/bluez-intro/x502.html) where on the server the socket towards the client is closed. Maybe the remote device keeps the Bluetooth connection open.

eelcocramer, 2017-08-04:
I've been reading through the docs and I do not think that the `shutdown` adds anything to this library. It is a more friendly way of bringing down a socket connection but in our case `close` should do the trick.

If someone finds anything that proves me otherwise I am going to assume that this library does everything that is possible to close the Bluetooth connection. I guess the behaviour described above is something that should be dealt with within the application code.

obermillerk, 2017-08-04:
I disagree with the ending there.  I think the behavior above should be handled in the library, because emitting a 'closed' event implies that the connection has been successfully closed, not that we told is to close.  I think that in the case that the close_socket call returns an error the library should determine what that error is and what exactly is means for the state of the device and handle it appropriately.  I'll gladly look into this a little later when I'm free.

eelcocramer, 2017-08-04:
I'm sorry for the misunderstanding. I agree, the library should be changed in such way that an error in the close is passed to the calling application. Though I'm still struggling about the options an application has to recover from such an error.

From the information in this thread I cannot determine if this error currently occurs an is a problem. You mentioned before that the connection is closed in good order, right?

obermillerk, 2017-08-05:
I didn't seem to get any errors on the disconnect, but what it rather seemed to be was that connecting would not function properly soon after a disconnect and I would say that this is a result of not being able to know if the connection was successfully closed or not and instead just assuming that it was closed.  I think having it wait for a successful close would at least improve if not solve this issue, though again something about the new hardware that I tested seemed to solve this issue we were having as well.  So in summary: We do not seem to be getting an error from the disconnect, but rather we are having issues I believe from the side effects of not waiting to see if the connection really closed or not.

obermillerk, 2017-08-05:
As for the errors, I'm reading a little more into this and I'll see if I can come up with any suggestions.

eelcocramer, 2017-08-07:
I've added the debug statement to the linux implementation on the `issue-162` branch so you can see if the `close` is actually successful or not.

wayofthefuture, 2017-08-07:
I'm not very versed with bindings or C, but I think the reason it is not connecting after is because it never disconnects. I have set a re-connect on a loop, and once disconnected in Linux it will never re-connect. The connect indicator light on the bluetooth device also indicates that a connection is still good and running. If it wasn't for the indicator light I wouldn't know the state of the connection. The instant I reset the device and terminate the connection, it connects right up again.

eelcocramer, 2017-08-07:
@wayofthefuture Can you use the `issue-162` branch and post your code + logging output?

If you call close using the the compiled code on this branch you should see logging that either the connection was closed successfully or that there was an error when closing the connection. In the latter case the logging should also note the error code.

obermillerk, 2017-08-14:
Here's my interpretations of the possible errors listed for the [`close_socket`](https://msdn.microsoft.com/en-us/library/windows/desktop/ms737582(v=vs.85).aspx) call:

* [WSANOTINITIALISED](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740668(v=vs.85).aspx#WSANOTINITIALISED) - This should (and probably is) already guaranteed to have been initialized in the code, so this should never happen unless there is something that needs to be fixed.
* [WSAENETDOWN](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740668(v=vs.85).aspx#WSAENETDOWN) - Something is wrong with the networking service, we shouldn't be connected anymore. This could be a little more complicated than I'm thinking it is, but it seems that it just means there is no networking capabilities at the moment, so we shouldn't have an active connection anymore.
* [WSAENOTSOCK](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740668(v=vs.85).aspx#WSAENOTSOCK) - Passed something that wasn't a socket. Again, this should already be guaranteed, so this shouldn't happen unless there's something to fix.
* [WSAEINPROGRESS](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740668(v=vs.85).aspx#WSAEINPROGRESS) - This is already in progress or we tried to cancel the disconnect but it already finished.  Since I don't believe we ever cancel the disconnect, this would mean that we're already disconnecting, so don't do anything, wait until we hear something else.
* [WSAEINTR](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740668(v=vs.85).aspx#WSAEINTR) - Our call was interrupted, but didn't complete. Still connected.
* [WSAEWOULDBLOCK](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740668(v=vs.85).aspx#WSAEWOULDBLOCK) - Our call was on a non-blocking socket, but couldn't be completed immediately, so it didn't happen at all, so we're still connected.

eelcocramer, 2017-08-26:
Thanks for doing this research. It is very helpful.

I will need help to implement this on Windows as I have no Windows development environment. Also test the implementation implementing this will need

obermillerk, 2017-09-01:
Hi,
Sorry this is so far back, but I was just looking at windows documentation for bluetooth to implement some pair/unpair stuff and I saw the shutdown command again and it made me think about this.  I don't recall if I ever tested the branch you made that uses this command but I'm looking at the documentation  (specifically on [this page](https://msdn.microsoft.com/en-us/library/windows/desktop/aa362909(v=vs.85).aspx)) and it looks like shutdown is whats used to disconnect.  I think the close_socket call is just used to free resources and doesn't actually tell the device that it's closing the connection (see [here](https://msdn.microsoft.com/en-us/library/windows/desktop/aa362906(v=vs.85).aspx)).

eelcocramer, 2017-09-05:
From [what I read](https://msdn.microsoft.com/en-us/library/windows/desktop/ms740481(v=vs.85).aspx) `shutdown` can be used to start a [graceful disconnection](https://msdn.microsoft.com/en-us/library/windows/desktop/ms738547(v=vs.85).aspx).

Currently the calling program should make sure that the connection can closed safely. I will not implement these graceful disconnects in this module myself but I will be happy to accept a PR that implements this.

jeffbk, 2018-03-20:
Hi.

First, let me just say thank you for creating and maintaining this important and useful library.  

I am using this library on Raspberry PI (Scratch) connecting to various custom bluetooth devices.  I am have a related issue with close() and have been investigating with the 162 branch. My test code connects to a device, waits 3 seconds, disconnects from the deivce, waits 3 seconds and then trys to  reconnect with the device (using a new btSerial object per #90).  After calling close on the btSerial, I receive the "closed" event, but the connection does not actually close.  There is no errror on close, as I see the extra logging output "**** close the BT connection was successful".  Here is my test code:
```
const BtSerial = require('bluetooth-serial-port');
const Logger = require('../lib/Logger');

const bluetoothAddress = "00:80:e1:ba:0f:30";

let btSerial = null;

let connecting = false;

function connect() {

    if (btSerial && btSerial.isOpen()) {
        return;
    }

    if (connecting) {
        return;
    }

    connecting = true;

    Logger.info("connecting");
    btSerial =  new BtSerial.BluetoothSerialPort();
    closed = false;
    btSerial.on('closed', function() {
        Logger.info("Closed");
        setTimeout(connect, 3000);
    });
    btSerial.on('error', function (err) {
        Logger.error(err);
        setTimeout(connect, 3000);
    });
    btSerial.on('data', function(data) {
        Logger.info(data);
    });

    btSerial.connect(bluetoothAddress, 1,
        function () {

        Logger.info("connected");
            setTimeout(disconnect, 3000);
            connecting = false;
        },
        function (err) {
            Logger.error(err);
            setTimeout(connect, 3000);
            connecting = false;
        });
}

function disconnect() {
    btSerial.close();
}

connect();  

```

I found that adding a call to shutdown(rfcomm->s, SHUT_RDWR) in BTSerialPortBinding.cc before the close() insures that the device is disconnected upon calling close and I am successfully able to reconnect to the device.  Without the shutdown() call, the connection does not close, and I cannot reconnect (connect attempt will fail immediately).

Let me know if you would like me to submit a pull request to implement this fix.

Thank you,
Jeff

jeffbk, 2018-03-20:
Hi,
I also wanted to comment on the close() issue.  I think it would be useful to the caller of close() to get an error if close() returns an error.  I think it could either be emitted as an 'error' event or perhaps the 'closed' event could contain an optional value that could indicate the error.  Otherwise, the error might occur and be hidden from the caller, while the caller thinks close() was successful.

Thank you,
Jeff

eelcocramer, 2018-03-21:
@jeffbk thanks a lot for your offer to create a PR,  I would be happy to consider it. 👍

eelcocramer, 2018-03-26:
@jeffbk added the shutdown to the linux code and I've just added the same change to the windows code. Both changes are currently available in the `next-release` branch.

jeffbk, 2018-03-26:
@eelcocramer Thank you for considering my PR.  I really appreciate your efforts maintaining this excellent and useful project.

eelcocramer, 2018-04-06:
Anyone willing to test the `next-release` branch on windows before I merge this into master? Maybe @obermillerk?

obermillerk, 2018-04-06:
I am back in classes and will likely not be able to test this any time soon, sorry

eelcocramer, 2018-04-06:
No problem. Thanks a lot for your response. 👍🏼

eelcocramer, 2018-04-12:
This should be fixed in release 2.1.6

closed 2018-04-12 by eelcocramer