eelcocramer/node-bluetooth-serial-port #163: 404 in install types for typescript

yangoff, 2017-08-07:
`kai-desktop BlueJS # npm install --save  @types/bluetooth-serial-port`
`npm ERR! code E404`
`npm ERR! 404 Not Found: @types/bluetooth-serial-port@latest`

meisterlampe, 2017-08-08:
The package itself contains the typescript definition file. No need to install them directly:

- Import the package: `import { BluetoothSerialPortServer } from 'bluetooth-serial-port';`
- Give your server variable a type: `let server: BluetoothSerialPortServer = new BluetoothSerialPortServer();`
- E.g. in VSCode type `server.` and let Intellisense guide you...

yangoff, 2017-08-08:
Uncaught TypeError: exists is not a function
    at Function.getRoot (bindings.js:153)
    at bindings (bindings.js:55)
    at bluetooth-serial-port.js:20
    at Object.<anonymous> (bluetooth-serial-port.js:137)
    at Object.../../../../bluetooth-serial-port/lib/bluetooth-serial-port.js (bluetooth-serial-port.js:240)
    at __webpack_require__ (bootstrap f88ba37…:54)
    at Object.../../../../../src/app/service/service.component.ts

./~/bindings/bindings.js
76:22-40 Critical dependency: the request of a dependency is an expression

eelcocramer, 2017-08-08:
Thank you for the report. To me this last comment seems not related to the original issue. If this is indeed the case I would appreciate if you can close this issue and create a new one. This keeps issues separated and easier to search for.

Also please add more details about your issue: what platform are you using, what version of nodejs, a code example that can be used to reproduce the issue, etc.

From the stacktrace I can see that this issue occurs in `bindings` which is not part of this project. I noticed that that a [new version](https://www.npmjs.com/package/bindings) of this module was released recently which changes the way the `exists` function is determined. It might solve your issue.

Thanks!

closed 2018-04-12 by eelcocramer