eelcocramer/node-bluetooth-serial-port #164: npm install bluetooth-serial-port error

rabdennebi, 2017-08-20:
hello, i have a problem with this installation. Can you help me please ?

npm ERR! 404 Not Found
npm ERR! 404
npm ERR! 404 'types/node' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it
npm ERR! 404 It was specified as a dependency of 'bluetooth-serial-port'
npm ERR! 404
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, or http url, or git url.

npm ERR! System Linux 4.9.41-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "bluetooth-serial-port" "--msvs_version=2013"
npm ERR! cwd /home/pi/Desktop/projetDrone
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! code E404
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Desktop/projetDrone/npm-debug.log
npm ERR! not ok code 0

rasberry pi 3  RASPBIAN STRETCH WITH DESKTOP

NervenCid, 2017-08-21:
I got the same problem with raspberry pi i get 

directory not found

#include <bluetooth/bluetooth.h>

eelcocramer, 2017-08-26:
`types/node` is an optional dependency so the installation should succeed when it is not available. Can you post your complete log including the command you use to install the module?

closed 2018-04-12 by eelcocramer