eelcocramer/node-bluetooth-serial-port #165: Installation failed on linux

NervenCid, 2017-08-22:
I tried to install on raspbian but i got this error instead:

 CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o
../src/linux/DeviceINQ.cc:35:37: fatal error: bluetooth/bluetooth.h: No existe el fichero o el directorio
     #include <bluetooth/bluetooth.h>
                                     ^
compilation terminated.
BluetoothSerialPort.target.mk:99: fallo en las instrucciones para el objetivo 'Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o'
make: *** [Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o] Error 1
make: se sale del directorio '/usr/lib/node_modules/bluetooth-serial-port/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.9.41-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build"
gyp ERR! cwd /usr/lib/node_modules/bluetooth-serial-port
gyp ERR! node -v v8.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-serial-port@2.1.4 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-serial-port@2.1.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-08-22T01_08_24_729Z-debug.log

eelcocramer, 2017-08-22:
I have to be brief but please check the installation instructions. I guess libbluetooth-dev is not installed.

NervenCid, 2017-08-23:
@eelcocramer thank you for the answer:

so shall install that lib on raspberry pi previously?

eelcocramer, 2017-08-26:
Sorry, but I do not understand your question.