eelcocramer/node-bluetooth-serial-port #169: macOS 10.13 Exception NSInvalidArgumentException

eelcocramer, 2017-11-08:
Due to breaking changes in the Bluetooth API

JackCurrie, 2017-11-20:
Been exploring bluetooth libraries for the last few days and trying to run them on High Sierra, looks like this update effected a few of them. [noble](https://github.com/sandeepmistry/noble) is a related library, and they fixed the issue with [this PR.](https://github.com/sandeepmistry/noble/pull/689/files)

A little too busy to dig into it, but I wanted to just put this up so that anybody else who is interested in helping might have somewhere to start looking!

gengshenghong, 2018-01-02:
@eelcocramer How did you see the exception message?

I am trying to look into this issue but I do not see the exception, using the "sync.js" to test inquiry, it does not return any results (it seems hang?). I also tried to start with lldb and also no crash/exception messages. Just want to confirm how you get the exception message before looking into more details.

[image]

Thanks,
Shenghong

eelcocramer, 2018-01-02:
I'm executing the same command:

```
> lldb -- node sync.js    20s  Tue Jan  2 13:04:39 2018
(lldb) target create "node"
Current executable set to 'node' (x86_64).
(lldb) settings set -- target.run-args  "sync.js"
(lldb) r
Process 11781 launched: '~/.nvm/versions/node/v6.11.1/bin/node' (x86_64)
start inquiry
2018-01-02 13:05:03.369226+0100 node[11781:443119] Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
2018-01-02 13:05:03.370879+0100 node[11781:443119] Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
```

When I use node v7.10.1 I get the same exception.

gengshenghong, 2018-01-03:
Interesting, I am using node v6.9.1 and I have no exceptions. After adding some logs around inquiry code, I do get exception, but at different places:
[image]

I'll take a look and hope to get some ideas...

eelcocramer, 2018-01-03:
... and a different exception, so this might be another issue. What version of MacOS are you running?

If I use node v6.9.1  I get the same result as I posted above.

gengshenghong, 2018-01-03:
I am using Mac OS X high Sierra, in a virtual machine (I do not want to upgrade my MacBook to high sierra before fixing the issue.)

The system image I got from Google Search:
https://saintlad.com/install-macos-high-sierra-in-virtualbox-on-windows-10/

gengshenghong, 2018-01-04:
@eelcocramer Found something interesting. I've looks at the details on your code and failed to find any API issues.
(API headers: https://github.com/phracker/MacOSX-SDKs/blob/master/MacOSX10.13.sdk/System/Library/Frameworks/IOBluetooth.framework/Versions/A/Headers/objc/IOBluetoothDeviceInquiry.h)

So, I decided to write a simple helloworld to test the behavior and found everything just works in my code, the code logic is just same with your libraries'. So, I got confused and I tried to copy the code into your library to verify and found if I put code into "inquireWithPipe" function, it works, if I put it into "inquiryTask", it does not work (I mean, no logs...), I use a simple delegate I wrote to test. It seems not an issue of Bluetooth APIs itself, maybe some other compatibility stuff causing the issue (threads?)??

The test delegate code:
[image]

The test code (if I put the same code in inquireWithPipe directly, I see all the logs in test delegate class):
[image]

It seems an issue of the woker/thread?

gengshenghong, 2018-01-04:
Below is the result when put the code in inquireWithPipe:
[image]

eelcocramer, 2018-01-04:
Maybe you can open a new issue for your finding? It seems unrelated to this issue. It will help to keep the discussion clean. Thanks.

gengshenghong, 2018-01-04:
Hmm, it is only reproducible in Mac OS High Sierra, I think the issue is same here from the test results (if confirmed to be thread-related, we may get different exception?). I'll check more details when I have time. If it is confirmed to be different issue, I'll re-submit a new one.

Thanks,
Shenghong

eelcocramer, 2018-01-04:
I see a similar behaviour under some circumstances. So sometimes I get the exception that started this issue, in other situation I notice the behaviour you encountered. I'm not yet sure why the behaviour is not always the same.

gengshenghong, 2018-01-04:
Great you see different results also. As thread is used, maybe it is possible to get random results. We may need to review the thread/worker code (maybe something changed in High Sierra?) to get the answer. :)

Thanks,
Shenghong

eelcocramer, 2018-01-08:
I noticed that only when a Bluetooth device is connected to my Mac I get the `NSInvalidArgumentException`. But the inquiry seems to hang indefinitely in any case.

eelcocramer, 2018-01-08:
So it looks like the runloop of `worker` thread is not started correctly but I did not find out why this was changed in High Sierra that might cause this behaviour.

gengshenghong, 2018-01-10:
Yes, even I suspect the worker is not started correctly (or maybe stopped directly) which make the inquiry does not work according to my test above. Maybe need to write some simple test code to test the behavior of the NSThread (not familiar at all, I am trying to learn) on High Sierra...

iboshkov, 2018-05-17:
@eelcocramer Any updates on this issue ? I am still able to reproduce this on High Sierra and the master branch. Did you ever investigate what's up ?

eelcocramer, 2018-05-17:
No updates. I don't see this error anymore on my configuration. There is an other error on node v10 but on v8 stuff works fine for me.

eelcocramer, 2018-05-17:
Oh wait. That is another issue. This was a runtime exception and I still see that in my configuration.

Still haven't investigate this any further though.

iboshkov, 2018-05-17:
Does the library work for you on a Mac ? On High Sierra and the examples (sync/async) both hang. I'm going to try to find some time and dive into the native code, seems like it's unavoidable in order to resolve this.

eelcocramer, 2018-05-17:
Don’t know. I haven’t tried it for a while. But from the comments in this thread I guess it isn’t working an the problem is most likely in the worker thread.

eelcocramer, 2019-08-23:
I haven't seen this issue for a while now. Closing for now.

closed 2019-08-23 by eelcocramer