eelcocramer/node-bluetooth-serial-port #173: Install fails on Windows 10

zavitax, 2018-01-25:
Install fails for me with:
> npm install bluetooth-serial-port

- Running from Developer Command Prompt for VS 2017
- Python accessible from command line:
> Python 2.7.6 (default, Nov 10 2013, 19:24:18) [MSC v.1500 32 bit (Intel)] on win32

0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'install',
1 verbose cli   'bluetooth-serial-port' ]
2 info using npm@5.6.0
3 info using node@v9.4.0
4 verbose npm-session 61f152799112b18e
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 http fetch GET 304 https://registry.npmjs.org/bluetooth-serial-port 421ms (from cache)
8 silly pacote tag manifest for bluetooth-serial-port@latest fetched in 489ms
9 silly install loadIdealTree
10 silly install cloneCurrentTreeToIdealTree
11 silly install loadShrinkwrap
12 silly install loadAllDepsIntoIdealTree
13 silly resolveWithNewModule bluetooth-serial-port@2.1.5 checking installable status
14 http fetch GET 304 https://registry.npmjs.org/bindings 84ms (from cache)
15 silly pacote range manifest for bindings@1.2.x fetched in 88ms
16 silly resolveWithNewModule bindings@1.2.1 checking installable status
17 http fetch GET 304 https://registry.npmjs.org/nan 235ms (from cache)
18 silly pacote tag manifest for nan@latest fetched in 239ms
19 silly resolveWithNewModule nan@2.8.0 checking installable status
20 http fetch GET 304 https://registry.npmjs.org/@types%2fnode 1007ms (from cache)
21 silly pacote range manifest for @types/node@^7.0.10 fetched in 1012ms
22 silly resolveWithNewModule @types/node@7.0.52 checking installable status
23 silly currentTree base64gen@1.0.0
23 silly currentTree `-- pngjs2@2.0.0
24 silly idealTree base64gen@1.0.0
24 silly idealTree +-- @types/node@7.0.52
24 silly idealTree +-- bindings@1.2.1
24 silly idealTree +-- bluetooth-serial-port@2.1.5
24 silly idealTree +-- nan@2.8.0
24 silly idealTree `-- pngjs2@2.0.0
25 silly install generateActionsToTake
26 silly diffTrees action count 4
27 silly diffTrees add @types/node@7.0.52
28 silly diffTrees add bindings@1.2.1
29 silly diffTrees add nan@2.8.0
30 silly diffTrees add bluetooth-serial-port@2.1.5
31 silly decomposeActions action count 32
32 silly decomposeActions fetch @types/node@7.0.52
33 silly decomposeActions extract @types/node@7.0.52
34 silly decomposeActions preinstall @types/node@7.0.52
35 silly decomposeActions build @types/node@7.0.52
36 silly decomposeActions install @types/node@7.0.52
37 silly decomposeActions postinstall @types/node@7.0.52
38 silly decomposeActions finalize @types/node@7.0.52
39 silly decomposeActions refresh-package-json @types/node@7.0.52
40 silly decomposeActions fetch bindings@1.2.1
41 silly decomposeActions extract bindings@1.2.1
42 silly decomposeActions preinstall bindings@1.2.1
43 silly decomposeActions build bindings@1.2.1
44 silly decomposeActions install bindings@1.2.1
45 silly decomposeActions postinstall bindings@1.2.1
46 silly decomposeActions finalize bindings@1.2.1
47 silly decomposeActions refresh-package-json bindings@1.2.1
48 silly decomposeActions fetch nan@2.8.0
49 silly decomposeActions extract nan@2.8.0
50 silly decomposeActions preinstall nan@2.8.0
51 silly decomposeActions build nan@2.8.0
52 silly decomposeActions install nan@2.8.0
53 silly decomposeActions postinstall nan@2.8.0
[... 92 lines cut ...]
142 silly saveTree | `-- nan@2.8.0
142 silly saveTree `-- pngjs2@2.0.0
143 warn base64gen@1.0.0 No description
144 warn base64gen@1.0.0 No repository field.
145 verbose stack Error: bluetooth-serial-port@2.1.5 install: `node-gyp configure build`
145 verbose stack Exit status 1
145 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:285:16)
145 verbose stack     at EventEmitter.emit (events.js:160:13)
145 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
145 verbose stack     at ChildProcess.emit (events.js:160:13)
145 verbose stack     at maybeClose (internal/child_process.js:943:16)
145 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
146 verbose pkgid bluetooth-serial-port@2.1.5
147 verbose cwd C:\Shared\TalkBack\Dev\Bluetooth
148 verbose Windows_NT 10.0.16299
149 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "bluetooth-serial-port"
150 verbose node v9.4.0
151 verbose npm  v5.6.0
152 error code ELIFECYCLE
153 error errno 1
154 error bluetooth-serial-port@2.1.5 install: `node-gyp configure build`
154 error Exit status 1
155 error Failed at the bluetooth-serial-port@2.1.5 install script.
155 error This is probably not a problem with npm. There is likely additional logging output above.
156 verbose exit [ 1, true ]

zavitax, 2018-01-25:
And here's the solution:
https://github.com/nodejs/node-gyp#installation

Namely:

Install all the required tools and configurations using Microsoft's windows-build-tools using npm install --global --production windows-build-tools from an elevated PowerShell or CMD.exe (run as Administrator).

zavitax, 2018-01-25:
Unfortunately, when doing this:

const BluetoothSerialPort = require('bluetooth-serial-port');

I get:

C:\Shared\TalkBack\Dev\Bluetooth>node index.js
C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bindings\bindings.js:83
        throw e
        ^

Error: %1 is not a valid Win32 application.
\\?\C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\Debug\BluetoothSerialPort.node
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at bindings (C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bindings\bindings.js:76:44)
    at C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:20:39
    at Object.<anonymous> (C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:137:2)

eelcocramer, 2018-01-25:
Does the file `C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\Debug\BluetoothSerialPort.node` exists?

In the FAQ labelled issues you can find information on common Windows installation issues.

zavitax, 2018-01-25:
The file C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node exists

Note \Release\ as opposed to \Debug\

eelcocramer, 2018-01-25:
There is your problem. It should be compiled into `Release` when you use `npm install`.

zavitax, 2018-01-25:
It *is* compiled into Release when I use npm install.

The thing is, that later I opened C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\binding.sln and produced a Debug build as well.

So we've got both C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node produced by npm install, *and* C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node

But the issue still exists.

eelcocramer, 2018-01-25:
Ok, understood. You might want to check if it is compiled as a 32 bits binary (like the error message indicates). You may also want to check if you have the problem with the LTS version of Node (currently v8) instead of v9.

zavitax, 2018-01-25:
It's compiled as 64-bit binary.

Can you describe how to check for the 2nd part?

zavitax, 2018-01-25:
After changing build configuration and recompiling as 32-bit (both Debug and Release), I get the following error message:

C:\Shared\TalkBack\Dev\Bluetooth>node index.js
C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bindings\bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\Debug\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\out\Debug
\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\Debug\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\out\Release\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\Release\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\build\default\BluetoothSerialPort.node
 → C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\compiled\6.11.1\win32\ia32\BluetoothSerialPort.node
    at bindings (C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bindings\bindings.js:88:9)
    at C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:20:39
    at Object.<anonymous> (C:\Shared\TalkBack\Dev\Bluetooth\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:137:2)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)

zavitax, 2018-01-25:
Hm... my apologies. The .node files never get built when platform target is switched to Win32.

Here's the build log for Release build:

1>------ Rebuild All started: Project: BluetoothSerialPort, Configuration: Release Win32 ------
1>Build started 2018-01-25 17:45:34.
1>Target _PrepareForClean:
1>  Deleting file "Release\obj\BluetoothSerialPort\Bluetoot.E57CA2BD.tlog\BluetoothSerialPort.lastbuildstate".
1>Target InitializeBuildStatus:
1>  Touching "Release\obj\BluetoothSerialPort\Bluetoot.E57CA2BD.tlog\unsuccessfulbuild".
1>Target ClCompile:
1>  BluetoothSerialPort.cc
1>  DeviceINQ.cc
1>  BTSerialPortBinding.cc
1>  BluetoothHelpers.cc
1>  win_delay_load_hook.cc
1>  ..\src\windows\BTSerialPortBinding.cc(190): warning C4996: 'v8::Function::NewInstance': was declared deprecated
1>  C:\Users\ilya7\.node-gyp\9.4.0\include\node\v8.h(3843): note: see declaration of 'v8::Function::NewInstance'
1>  ..\src\windows\DeviceINQ.cc(182): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>  ..\src\windows\DeviceINQ.cc(182): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
1>  ..\src\windows\DeviceINQ.cc(182): note: the missing variadic argument 2 is required by format string '%18[^)]'
1>  ..\src\windows\DeviceINQ.cc(182): note: this argument is used as a buffer size
1>  ..\src\windows\DeviceINQ.cc(412): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>  ..\src\windows\DeviceINQ.cc(412): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
1>  ..\src\windows\DeviceINQ.cc(412): note: the missing variadic argument 2 is required by format string '%18[^)]'
1>  ..\src\windows\DeviceINQ.cc(412): note: this argument is used as a buffer size
1>  ..\src\windows\DeviceINQ.cc(413): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>  ..\src\windows\DeviceINQ.cc(413): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
1>  ..\src\windows\DeviceINQ.cc(413): note: the missing variadic argument 2 is required by format string '%18[^)]'
1>  ..\src\windows\DeviceINQ.cc(413): note: this argument is used as a buffer size
1>Done building target "ClCompile" in project "BluetoothSerialPort.vcxproj".
1>Target Link:
1>  node.lib(node.exe) : fatal error LNK1190: invalid fixup found, type 0x0003
1>Done building target "Link" in project "BluetoothSerialPort.vcxproj" -- FAILED.
1>
1>Done building project "BluetoothSerialPort.vcxproj" -- FAILED.
1>
1>Build FAILED.
1>
1>..\src\windows\BTSerialPortBinding.cc(190): warning C4996: 'v8::Function::NewInstance': was declared deprecated
1>..\src\windows\DeviceINQ.cc(182): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>..\src\windows\DeviceINQ.cc(412): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>..\src\windows\DeviceINQ.cc(413): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>node.lib(node.exe) : fatal error LNK1190: invalid fixup found, type 0x0003
1>    4 Warning(s)
1>    1 Error(s)
1>
1>Time Elapsed 00:00:04.33
========== Rebuild All: 0 succeeded, 1 failed, 0 skipped ==========

Here's the same for Debug build:

1>------ Rebuild All started: Project: BluetoothSerialPort, Configuration: Debug Win32 ------
1>Build started 2018-01-25 17:46:40.
1>Target _PrepareForClean:
1>  Deleting file "Debug\obj\BluetoothSerialPort\Bluetoot.E57CA2BD.tlog\BluetoothSerialPort.lastbuildstate".
1>Target InitializeBuildStatus:
1>  Touching "Debug\obj\BluetoothSerialPort\Bluetoot.E57CA2BD.tlog\unsuccessfulbuild".
1>Target ClCompile:
1>  BluetoothSerialPort.cc
1>  DeviceINQ.cc
1>  BTSerialPortBinding.cc
1>  BluetoothHelpers.cc
1>  win_delay_load_hook.cc
1>  ..\src\windows\DeviceINQ.cc(182): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>  ..\src\windows\DeviceINQ.cc(182): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
1>  ..\src\windows\DeviceINQ.cc(182): note: the missing variadic argument 2 is required by format string '%18[^)]'
1>  ..\src\windows\DeviceINQ.cc(182): note: this argument is used as a buffer size
1>  ..\src\windows\DeviceINQ.cc(412): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>  ..\src\windows\DeviceINQ.cc(412): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
[... 2 lines cut ...]
1>  ..\src\windows\DeviceINQ.cc(413): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>  ..\src\windows\DeviceINQ.cc(413): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
1>  ..\src\windows\DeviceINQ.cc(413): note: the missing variadic argument 2 is required by format string '%18[^)]'
1>  ..\src\windows\DeviceINQ.cc(413): note: this argument is used as a buffer size
1>  ..\src\windows\BTSerialPortBinding.cc(190): warning C4996: 'v8::Function::NewInstance': was declared deprecated
1>  C:\Users\ilya7\.node-gyp\9.4.0\include\node\v8.h(3843): note: see declaration of 'v8::Function::NewInstance'
1>Done building target "ClCompile" in project "BluetoothSerialPort.vcxproj".
1>Target Link:
1>  node.lib(node.exe) : fatal error LNK1190: invalid fixup found, type 0x0003
1>Done building target "Link" in project "BluetoothSerialPort.vcxproj" -- FAILED.
1>
1>Done building project "BluetoothSerialPort.vcxproj" -- FAILED.
1>
1>Build FAILED.
1>
[... cut ...]
1>..\src\windows\DeviceINQ.cc(412): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>..\src\windows\DeviceINQ.cc(413): warning C4473: 'sscanf_s' : not enough arguments passed for format string
1>..\src\windows\BTSerialPortBinding.cc(190): warning C4996: 'v8::Function::NewInstance': was declared deprecated
1>node.lib(node.exe) : fatal error LNK1190: invalid fixup found, type 0x0003
1>    4 Warning(s)
1>    1 Error(s)
1>
1>Time Elapsed 00:00:04.20
========== Rebuild All: 0 succeeded, 1 failed, 0 skipped ==========

eelcocramer, 2018-01-25:
The thing is that the windows support is community contributed and I cannot test it myself. If I look at the automatic build script we use `appveyor.yml` I notice that it uses Visual Studio 2015. Maybe VS2017 introduces a problem?

eelcocramer, 2018-01-25:
Maybe the logs from the automated builds are helpful: https://ci.appveyor.com/project/eelcocramer/node-bluetooth-serial-port

zavitax, 2018-01-25:
Tried npm install bluetooth-serial-port with VS2015 developer command prompt.

Got this:

C:\Shared\TalkBack\Dev\BluetoothVS2015>node index.js
C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bindings\bindings.js:83
        throw e
        ^

Error: %1 is not a valid Win32 application.
\\?\C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at bindings (C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bindings\bindings.js:76:44)
    at C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:20:39
    at Object.<anonymous> (C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:137:2)

Here's the log:

C:\Shared\TalkBack\Dev\BluetoothVS2015>npm install bluetooth-serial-port

> bluetooth-serial-port@2.1.5 install C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bluetooth-serial-port
> node-gyp configure build

C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bluetooth-serial-port>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure build )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" configure build )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  BluetoothSerialPort.cc
  DeviceINQ.cc
  BTSerialPortBinding.cc
  BluetoothHelpers.cc
  win_delay_load_hook.cc
..\src\windows\DeviceINQ.cc(44): warning C4996: '_WSAQuerySetA::dwSize': Use WS
AQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable depreca
ted API warnings [C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\bluetooth
-serial-port\build\BluetoothSerialPort.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1329): note: se
  e declaration of '_WSAQuerySetA::dwSize'
..\src\windows\DeviceINQ.cc(45): warning C4996: '_WSAQuerySetA::dwNameSpace': U
se WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable de
precated API warnings [C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\blue
tooth-serial-port\build\BluetoothSerialPort.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1334): note: se
  e declaration of '_WSAQuerySetA::dwNameSpace'
..\src\windows\DeviceINQ.cc(46): warning C4996: '_WSAQuerySetA::lpServiceClassI
d': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disab
le deprecated API warnings [C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules
\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1331): note: se
  e declaration of '_WSAQuerySetA::lpServiceClassId'
..\src\windows\DeviceINQ.cc(47): warning C4996: '_WSAQuerySetA::dwNumberOfCsAdd
rs': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disa
ble deprecated API warnings [C:\Shared\TalkBack\Dev\BluetoothVS2015\node_module
s\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1340): note: se
  e declaration of '_WSAQuerySetA::dwNumberOfCsAddrs'
..\src\windows\DeviceINQ.cc(48): warning C4996: '_WSAQuerySetA::lpszContext': U
se WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable de
precated API warnings [C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\blue
tooth-serial-port\build\BluetoothSerialPort.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1336): note: se
  e declaration of '_WSAQuerySetA::lpszContext'
..\src\windows\DeviceINQ.cc(61): warning C4996: '_WSAQuerySetA::lpcsaBuffer': U
se WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable de
precated API warnings [C:\Shared\TalkBack\Dev\BluetoothVS2015\node_modules\blue
tooth-serial-port\build\BluetoothSerialPort.vcxproj]
[... 205 lines cut ...]
  BluetoothSerialPort.exp
  Generating code
  All 252 functions were compiled because no usable IPDB/IOBJ from previous com
  pilation was found.
  Finished generating code
  BluetoothSerialPort.vcxproj -> C:\Shared\TalkBack\Dev\BluetoothVS2015\node_mo
  dules\bluetooth-serial-port\build\Release\\BluetoothSerialPort.node
  BluetoothSerialPort.vcxproj -> C:\Shared\TalkBack\Dev\BluetoothVS2015\node_mo
  dules\bluetooth-serial-port\build\Release\BluetoothSerialPort.pdb (Full PDB)
  win_delay_load_hook.cc
  Generating code
  All 1 functions were compiled because no usable IPDB/IOBJ from previous compi
  lation was found.
  Finished generating code
  BluetoothSerialPortServer.vcxproj -> C:\Shared\TalkBack\Dev\BluetoothVS2015\n
  ode_modules\bluetooth-serial-port\build\Release\\BluetoothSerialPortServer.no
  de
  BluetoothSerialPortServer.vcxproj -> C:\Shared\TalkBack\Dev\BluetoothVS2015\n
  ode_modules\bluetooth-serial-port\build\Release\BluetoothSerialPortServer.pdb
   (Full PDB)
npm WARN bluetoothcontroller@1.0.0 No description
npm WARN bluetoothcontroller@1.0.0 No repository field.

+ bluetooth-serial-port@2.1.5
updated 1 package in 12.047s

zavitax, 2018-01-25:
Tried building with VS2015, configuring Win32 platform.

Got this:

https://msdn.microsoft.com/query/dev14.query?appId=Dev14IDEF1&l=EN-US&k=k(LNK1112)&rd=true

LNK1112: module machine type 'x64' conflicts with target machine type 'X86'
Project: BluetoothSerialPort
File: node.lib (node.exe)
Line: 1

Perhaps this supports only 32-bit versions of node.exe ?

eelcocramer, 2018-01-26:
Maybe. The automatic build seems to work with both 32 as 64 bits versions but it looks like the module compiles to 32 bits.

eelcocramer, 2018-01-26:
I've changed the automated build configuration to include more recent versions of node and to use VS2017. On the build server everything seems to build fine: https://ci.appveyor.com/project/eelcocramer/node-bluetooth-serial-port/build/1.0.177

I've created an issue (#174) to included pre-build binaries in published versions of this module. That may be more friendly to users. Do not expect this to be available soon though.

closed 2020-02-06 by eelcocramer