eelcocramer/node-bluetooth-serial-port #174: Distribute binaries

eelcocramer, 2018-01-26:
Currently to install this module you need a build environment present on the machine you want to install the module on.

It would be better and more user friendly to distribute the binaries that are build during automated builds.

See also: https://github.com/node-serialport/node-serialport/blob/master/PUBLISHING.md

#173

wayofthefuture, 2018-05-19:
I understand this is to auto build binaries using "pre-build" but that doesn't seem realistic to me if it takes 1 hour to download the binaries. Is there any small version of the binaries? The visual studio library for windows is approx 500MB, is there another set of files we can use or a partial selection of files we could extract from visual studio to decrease the size of the download?

Then, at that point, we could implement something like above which would automate everything. I think first though, we would have to find an alternative source of a much smaller size. Thanks...

eelcocramer, 2018-05-19:
The prebuild binaries are tiny compared to visual studio. You can check their size in the build folder after install. I think the suffix is ‘.node’

wayofthefuture, 2018-05-20:
So I guess maybe if I can isolate these files, I could put a script in node "prebuild" to move them to the appropriate place before installing?

eelcocramer, 2018-05-20:
Sort off. It just something that needs to be configured but its currently not on my priority list.

wayofthefuture, 2018-05-21:
It seems that the large dependency is actually a dependency of Node-GYP, so there is really no way to pluck the small files for a build. I think at this point, I will have to locate the ".node" files and copy them into the repository. Then at that point, remove the node-gyp dependency and the build.

Obviously, I'd have to fork this repo and publish as something like: node-bluetooth-serial-port-windows, which would already be prebuilt for windows. This way, the dependency would install in seconds and wouldn't require such a huge download.

Do you see any problems with this? Would the ".node" file work on various Windows 10 computers, or would something as simple as a chipset difference break the build? Thanks...

eelcocramer, 2018-05-22:
Sure, that would work but I see numerous problems.

First of all it is just annoying and error prone to have to manually do this every release. It is not the wayofthefuture :-) (pun intended)

But the real problem is thrust and accountability. When a continues integration script build and releases the binaries the binaries can be tracked to a specific commit and a build server like travis or appveyor acts like a trusted 3rd party that builds the binary on well monitored infrastructure. These parties make sure that the binaries are virus / malware free.

wayofthefuture, 2018-05-22:
Do the CI servers have the ability to build Windows binaries which would then be compatible with Windows operating systems?

eelcocramer, 2018-05-22:
Yes this is already done after each commit to this repository. All platforms are being build against a selected set of nodejs to check whether the commit introduced a build error. See also the images in the top of the readme.