eelcocramer/node-bluetooth-serial-port #176: Problems with "Basic client usage" example

RaRaton, 2018-02-23:
Hi all!
First, thanks for this module it is really good that you shared it.  I would like to use it to communicate with my Arduino Uno. 

- I have node.js  v8.9.4, npm 5.6.0 and python 2.7.14. installed.
- I installed/downloaded your module using: **npm install bluetooth-serial-port.**
- I am running the example in a command console of windows 10.

I tested your examples and all are working fine. So, I think I have all well installed, although I got several warnings during the download of the module.

I started using your Basic client usage example with few small variations:

```
var BTSP = require('bluetooth-serial-port');
var serial = new BTSP.BluetoothSerialPort();
 
serial.on('found', function(address, name) {
 
    // you might want to check the found address with the address of your
    // bluetooth enabled Arduino device here.
    if (name != "NEW_PCB_V5") {
        console.log("Device " + name + " is no our device");
        return;
    }        

    var bluetoothAddress = address;
    console.log("Device " + name + " Bluetooth address: " + bluetoothAddress);

    serial.findSerialPortChannel(address, function (channel) {

        console.log('Found RFCOMM channel for serial port on ' + name + ': ' + channel);

        console.log('Attempting to connect...');

        serial.connect(bluetoothAddress, channel, function () {

            console.log('Connected');
            
            process.stdin.resume();
            process.stdin.setEncoding('utf8');
            console.log('Press "0" or "1" and "ENTER" to get value')
 
            process.stdin.on('data', function (data) {
                console.log("Request: " + data);
                serial.write(data);                
            });
 
            serial.on('data', function(data) {
                console.log('Received: ' + data);
            });
        }, function () {
            console.log('cannot connect');
        });
    });
});
 
serial.on('finished', function () {
    console.log('scan did finish');
});

serial.inquire();
```

Also, I changed a little your .ino file to avoid the use of a led. I have tested using Arduino IDE and it is working fine. 

```
// This code uses the SoftwareSerial library.
// It can be obtained here: http://arduino.cc/en/Reference/SoftwareSerial

unsigned int timeout=0;
unsigned char state=0;

char currentVal;
char val;         // variable to receive data from the serial port

// Timer2 service
ISR(TIMER2_OVF_vect) { 

  TCNT2 = 0;
  timeout++;

  if (timeout>61) {
    state=1;
    timeout=0;
  }

}

// initialize the timer 2 service

void init_timer2(void) {

  TCCR2A |= (1 << WGM21) | (1 << WGM20);   

  TCCR2B |= 0x07;                         // by clk/1024

  ASSR |= (0<<AS2);                       // Use internal clock - external clock not used in Arduino

  TIMSK2 |= 0x01;                         //Timer2 Overflow Interrupt Enable

  TCNT2 = 0;

  sei();   

}

// sets up the program

void setup() {

  // open the serial port

  Serial.begin(9600);

  // interrupt for reading from the bluetooth connection 

  attachInterrupt(0, cleantime, FALLING);

  init_timer2();
}

// function for controlling the led

void control(void) {

  if (Serial.available()) {               // if data is available to read
    val = Serial.read();                  // read it and store it in 'val'
  }

  if (val == '1') {                       // if '1' was received

    Serial.println('1');                  // display the new value
    currentVal = val;

  } else if (val == '0') { 

    Serial.println('0');                  // display the new value
    currentVal = val;

  } else if (val == 's') {                // if 's' is received display the current status of the led

    if (currentVal == '1') {
      Serial.println('1');
    } else {
      Serial.println('0');
    } 
  }

  val = ' ';
  delay(100);                             // wait 100ms for next reading
}

// control loop for the program
void loop() {
  switch(state) {
    case 0:
      // no bt connection, do nothing
      break;

    case 1:
      // when there is a bt connection enter the control function
      control(); 
      break;
  }
}

void cleantime() {
  timeout=0;
  state=0;
}
```

My problem it is that when I run my example, arduino-example.js, I am able to connect with my device, select the value that I want to send to my Arduino but then I get and error. 

This is the full answer:

```
C:\Users\admin\Documents\experimental\node_modules\bluetooth-serial-port\examples>node arduino-example.js
Device PCB_V4_1 is no our device
Device NEW_PCB_V5 Bluetooth address: 98:D3:32:20:C6:4C
scan did finish
Found RFCOMM channel for serial port on NEW_PCB_V5: 1
Attempting to connect...
Connected
Press "0" or "1" and "ENTER" to get value.
0
Request: 0

Command Prompt - node  arduino-example.js[27948]: src\node_buffer.cc:218: Assertion `obj->IsArrayBufferView()' failed.
 1: node::DecodeWrite
 2: node::Buffer::Data
 3: BTSerialPortBinding::Write
 4: Nan::imp::FunctionCallbackWrapper
 5: v8::internal::CompilerDispatcherTracer::Estimate
 6: v8::internal::CompilerDispatcher::RemoveJob
 7: v8::internal::CompilerDispatcher::RemoveJob
 8: v8::internal::CompilerDispatcher::RemoveJob
 9: 0000030851C847A1
```

I looked into the code of node_buffer.cc and all files of the module but I am not sure what it is failing. I think it is the DecodeWrite function but I don't understand why.
Any advice or idea what is the problem? Also, can I debug the code?
I am new with node.js, phyton... so any reference to understand the error will be great! I want to learn more :)
Many thanks, 
Raquel

eelcocramer, 2018-04-12:
Hi @RaRaton, my excuses for this late response. Do you still have this issue?

codeiain, 2018-10-20:
Am having the same problem on my mac using the basic example

eelcocramer, 2018-10-20:
Can you try writing a `Buffer` instead of a string to see if this resolves the issue?

eelcocramer, 2018-10-23:
>             process.stdin.on('data', function (data) {
>                 console.log("Request: " + data);
>                 serial.write(data);                
>             });

A quick [check of the docs](https://nodejs.org/api/stream.html#stream_event_data) learns me that in the case above `data` can be either a `String` or a `Buffer`. `write` need the parameter to be a `Buffer` so please make sure that this is the case:

```
serial.write(Buffer.from(data));
```

Maybe a more helpful error would be better. Maybe `write` should check whether the parameter is a `Buffer` or not before continuing.

eelcocramer, 2018-10-23:
I  checked the example code but I believe they all write instances of ` Buffer` if I missed this somewhere please let me know and I will update the example.

closed 2018-10-23 by eelcocramer