eelcocramer/node-bluetooth-serial-port #180: Thoughts of updating the API

eelcocramer, 2018-04-12:
It may be a good idea to update the API to something more modern.

I noticed that [someone]() copied (not forked) the native code of this project and created an API what IMHO might be better suitable than the current state.

Would this be something that folks would like for version 2.0?

jdpigeon, 2018-04-19:
Took a look at that new project... looks alright, but I don't think it's that much of an improvement.

Also, I can say that I am just about to switch jobs and leave an app that relies on this library as a legacy project. Wouldn't want to have to run back and change things! :stuck_out_tongue:

wayofthefuture, 2018-05-18:
If it works and is reliable then why change it? I do think however the verbose output should be removed when "npm" installing the library...

eelcocramer, 2018-05-18:
Thanks for the feedback.

I guess there are 2 things that can be done about the verbose output. The first is to set the compiler options to ignore the warnings, the second is to fix the warnings. The latter is ofcourse better but is also more work to get it done.

wayofthefuture, 2018-05-19:
If the warnings really don't matter, then I think the former :)

closed 2020-02-06 by eelcocramer