eelcocramer/node-bluetooth-serial-port #182: Windows 10 installation failing with error C2661

MichaelMauderer, 2018-05-10:
After running `npm install bluetooth-serial-port` I'm seeing the following error: 

```
..\src\windows\BTSerialPortBinding.cc(188): error C2661: 'v8::Function::NewInstance': no overloaded function takes 2 arguments [C:\Users\######>\Desktop\######>\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
```

This is the full log  [Uploading 2018-05-10T14_35_17_981Z-debug.log…]()
This is he full command line output

```
C:\Users\######\Desktop\######>npm install bluetooth-serial-port

> bluetooth-serial-port@2.1.6 install C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port
> node-gyp configure build

C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure build )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" configure build )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  BluetoothSerialPort.cc
  DeviceINQ.cc
  BTSerialPortBinding.cc
  BluetoothHelpers.cc
  win_delay_load_hook.cc
..\src\windows\DeviceINQ.cc(104): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxpr
oj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\DeviceINQ.cc(180): warning C4473: 'sscanf_s' : not enough arguments passed for format string [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSeri
alPort.vcxproj]
  ..\src\windows\DeviceINQ.cc(180): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
  ..\src\windows\DeviceINQ.cc(180): note: the missing variadic argument 2 is required by format string '%18[^)]'
  ..\src\windows\DeviceINQ.cc(180): note: this argument is used as a buffer size
..\src\windows\DeviceINQ.cc(267): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxpr
oj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\DeviceINQ.cc(270): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxpr
oj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\DeviceINQ.cc(299): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxpr
oj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\DeviceINQ.cc(302): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxpr
oj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\DeviceINQ.cc(410): warning C4473: 'sscanf_s' : not enough arguments passed for format string [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSeri
alPort.vcxproj]
  ..\src\windows\DeviceINQ.cc(410): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
  ..\src\windows\DeviceINQ.cc(410): note: the missing variadic argument 2 is required by format string '%18[^)]'
  ..\src\windows\DeviceINQ.cc(410): note: this argument is used as a buffer size
..\src\windows\DeviceINQ.cc(411): warning C4473: 'sscanf_s' : not enough arguments passed for format string [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSeri
alPort.vcxproj]
  ..\src\windows\DeviceINQ.cc(411): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided
  ..\src\windows\DeviceINQ.cc(411): note: the missing variadic argument 2 is required by format string '%18[^)]'
  ..\src\windows\DeviceINQ.cc(411): note: this argument is used as a buffer size
..\src\windows\BTSerialPortBinding.cc(65): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerialP
ort.vcxproj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\BTSerialPortBinding.cc(74): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerialP
ort.vcxproj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\BTSerialPortBinding.cc(125): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerial
Port.vcxproj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\BTSerialPortBinding.cc(188): error C2661: 'v8::Function::NewInstance': no overloaded function takes 2 arguments [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\
build\BluetoothSerialPort.vcxproj]
..\src\windows\BTSerialPortBinding.cc(195): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerial
Port.vcxproj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\windows\BTSerialPortBinding.cc(351): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\BluetoothSerial
Port.vcxproj]
  C:\Users\######\Desktop\######\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
  win_delay_load_hook.cc
  Generating code
  All 1 functions were compiled because no usable IPDB/IOBJ from previous compilation was found.
  Finished generating code
  BluetoothSerialPortServer.vcxproj -> C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\Release\\BluetoothSerialPortServer.node
  BluetoothSerialPortServer.vcxproj -> C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPortServer.pdb (Full PDB)
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:235:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "build"
gyp ERR! cwd C:\Users\######\Desktop\######\node_modules\bluetooth-serial-port
gyp ERR! node -v v10.1.0
gyp ERR! node-gyp -v v3.6.2
```

I have tried running install with both `-msvs_version=2015` and `-msvs_version=2017` as well as specifying `--build-from-source` but this leads to the same error..

eelcocramer, 2018-05-10:
Can you try building with node v9?

eelcocramer, 2018-05-10:
Currently node v8 is the recommended LTS version of nodejs. v10 breaks the build on all platforms (for different reasons).

closed 2018-05-10 by eelcocramer

MichaelMauderer, 2018-05-11:
Thanks, downgrading to 8.11.1 did fix the issue.

eelcocramer, 2018-07-06:
Support for v10 has not been added. #189