eelcocramer/node-bluetooth-serial-port #183: can't send long commands with <CR> + <LF>

oumartraore, 2018-05-14:
Hi all!
First, thanks for this module it is really good that you shared it. 

I use ascii protocole in some project  to send some command like 'rb01'. But it not working, i think your librarie send buffer caractere by caractere.
How to make it possible to send all command, like we use putty or tera-term with <CR>+<LF> setting ?

eelcocramer, 2018-05-15:
Multiple bytes can be send within a single packet but there is no guarantee that the complete buffer is send in a single packet.

eelcocramer, 2018-05-15:
That said, if you are sure that a command would fit with the MTU of a packet it should be possible to send transfer the command in a whole. I'm not sure if the current code will guarantee this though, I would need to check the code for each platform to be sure.

oumartraore, 2018-05-16:
How to specify the size of my buffer, to allow the sending of all my buffer in a single packet ? I am on windows. Even if it's in code c

eelcocramer, 2018-05-16:
This is where the buffer is written on windows: https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/windows/BTSerialPortBinding.cc#L98

As for the other question: https://nodejs.org/api/events.html#events_emitter_removelistener_eventname_listener

oumartraore, 2018-05-16:
I thank you for your answer. I found the solution to my problem.

As you said, multiple bytes can be send, but there is no guarantee that the entire buffer is sent in a packet.

I used another library for the "serial port" able to send the data in "string" or in a "buffer".

I think that if we communicate through a protocol, it is more judicious to send a "string" than a "buffer".

this could help another newbie. I thank you for helping me

eelcocramer, 2018-05-16:
Thanks you for the hint. I've added the issue to tag FAQ.

closed 2018-05-16 by eelcocramer