eelcocramer/node-bluetooth-serial-port #186: program crashes if no Bluetooth available

zhaiyuzhai, 2018-05-28:
platform：win10_64bit;
app:electron1.7.9;
error:Process finished with exit code -1073741819 (0xC0000005)

My application is based on electron. When there is no proper bluetooth, the module will cause my app to crash。Someone to solve this problem, thank you very much？

closed 2018-05-28 by eelcocramer

zhaiyuzhai, 2018-05-29:
When I call btSerial.inquireSync().What seems to be blocking my app for about three seconds.
I still don't have a good solution to it

eelcocramer, 2018-05-29:
If you don’t want a synchronized call you should use inquire which works async.

zhaiyuzhai, 2018-05-29:
@eelcocramer Thank you very much for your help, but now I have a difficulty that when bluetooth is not available, the program will report an error and quit directly. As shown below.
Process finished with exit code -1073741819 (0xC0000005)；
I have seen#136， use try catch statements around btSerial.inquire() or btSerial.inquireSync() cannot work. I dont know how to catch the error  ，how to stop the program from crashing。

zhaiyuzhai, 2018-05-29:
@eelcocramer Bluetooth-serial-port donot has error Event，Hopefully you can add it in the next release