eelcocramer/node-bluetooth-serial-port #192: Segfault

darkrift, 2018-08-03:
I am trying to install this on ubuntu bionic with bluez 5.48. However, whenever I try to launch my program, it segfaults.

Here's the stacktrace using lldb : 

```
rlavoie@richard-laptop:~/work/rlavoie/blueanatomy$ lldb -- node index.js 
(lldb) target create "node"
Current executable set to 'node' (x86_64).
(lldb) settings set -- target.run-args  "index.js"
(lldb) r
Process 3791 launched: '/usr/bin/node' (x86_64)
123
234
Process 3791 stopped
* thread #7, name = 'node', stop reason = signal SIGSEGV: invalid address (fault address: 0x59b8)
    frame #0: 0x0000000000ac2bd0 node`v8::HandleScope::Initialize(v8::Isolate*) + 80
node`v8::HandleScope::Initialize:
->  0xac2bd0 <+80>: movq   0x59b8(%rbx), %r13
    0xac2bd7 <+87>: callq  0xf5fba0                  ; v8::internal::ThreadId::GetCurrentThreadId()
    0xac2bdc <+92>: movl   %eax, -0x30(%rbp)
    0xac2bdf <+95>: movl   0x28(%r13), %edx
(lldb) bt
* thread #7, name = 'node', stop reason = signal SIGSEGV: invalid address (fault address: 0x59b8)
  * frame #0: 0x0000000000ac2bd0 node`v8::HandleScope::Initialize(v8::Isolate*) + 80
    frame #1: 0x00007ffff4866c8d BluetoothSerialPort.node`Nan::HandleScope::HandleScope(this=0x00007fffe7ffecc0) at nan.h:344
    frame #2: 0x00007ffff4866f29 BluetoothSerialPort.node`Nan::ThrowError(msg="opening socket") at nan.h:769
    frame #3: 0x00007ffff4865c3a BluetoothSerialPort.node`DeviceINQ::doInquire() at DeviceINQ.cc:184
    frame #4: 0x00007ffff4867f06 BluetoothSerialPort.node`InquireWorker::Execute(this=0x00000000024a11c0) at DeviceINQ.cc:270
    frame #5: 0x00007ffff4867837 BluetoothSerialPort.node`Nan::AsyncExecute(req=0x00000000024a11c8) at nan.h:2153
    frame #6: 0x00000000009ee131 node`worker(arg=<unavailable>) at threadpool.c:83
    frame #7: 0x00007ffff6e6d6db libpthread.so.0`start_thread + 219
    frame #8: 0x00007ffff6b9688f libc.so.6`__GI___clone at clone.S:95
```

darkrift, 2018-08-03:
Ok, I got it to run by investigating the code and realizing the problem was that my bluetooth stack couldn't have a socket fd. I changed my bluetooth dongle and it fixed it, but the segfault is pretty hard on the user instead of a normal error.

darkrift, 2018-08-03:
I tried with both node 8.11 and 10.8

eelcocramer, 2018-08-03:
Thank you for the report. I’m glad you figured out a solution. There should definitly be a handler for that error.