eelcocramer/node-bluetooth-serial-port #195: Multiple connections

smelliott100, 2018-12-17:
I want to connect to multiple bluetooth devices at one time and listen for data.  I am using some echo buttons for a game I wrote.  I had everything running with two devices with the exception of closing the devices on exit (more than one close() hangs the node app.  After I added button three and four, express web page in the same app takes for ever to come up and hangs.  I read there was some discussion on enabling multiple devices, was it ever implemented?

eelcocramer, 2018-12-18:
Does your question relate to the server or the client?

Can you add a reference to the issue that hold the discussion you are refering too?

smelliott100, 2018-12-18:
#86, #48, #3.  Reading these, I have assumed you would have multiple connections like this:

const BluetoothSerial = require('bluetooth-serial-port')
var btSerial = new BluetoothSerial.BluetoothSerialPort();
var btSerial1 = new BluetoothSerial.BluetoothSerialPort();
var btSerial2 = new BluetoothSerial.BluetoothSerialPort();

The reason is I cannot figure out how to have more than one connection per btSerial as referenced above.  

When I use it as described above, three work great.  When I add the fourth, my mac has crashed and node express running in the same app stops working. 

Any suggestions?

eelcocramer, 2018-12-19:
Can you post a stack trace?

Like mentioned in the issues you are refen ing I cannot think of a reason why multiple connection are not possible but I never used the library in this way myself.

If the stack trace does not have any pointers you may try to run you program in gdb and compile the lib with debug flags.

closed 2020-02-06 by eelcocramer