eelcocramer/node-bluetooth-serial-port #203: Connection problem - Raspberry to temperature/water depth sensor (question)

max-ci, 2019-02-10:
It's rather a question, not an issue with the library. I hope someone can help me with my problem.

I have Raspberry and I'm trying to connect to a Bluetooth Device (sensor for measuring temperature and water depth) and read data from it.

I'm able to pair the device with Raspberry, you can see that on the screenshot (bluetoothctl).

[image]

Inside that sensor there is IS1678S chip.
http://ww1.microchip.com/downloads/en/DeviceDoc/IS1678S_UARTCommandSet_UserGuide.pdf

My code:

```js
var btSerial = new (require('bluetooth-serial-port')).BluetoothSerialPort();

btSerial.on('found', function(address, name) {
  if(name === 'Fish Helper') {
    console.log(btSerial.isOpen())
    btSerial.findSerialPortChannel(address, function(channel) {
      console.log(channel);
      btSerial.connect(address, channel, function() {
        console.log('connected');

        btSerial.write(Buffer.from('my data', 'utf-8'), function(err, bytesWritten) {
          if (err) console.log(err);
        });

        btSerial.on('data', function(buffer) {
          console.log(buffer.toString('utf-8'));
        });
      }, function (e) {
        console.log(e); // Cannot connect error HERE
      });

      // close the connection when you're ready
      btSerial.close();
    }, function() {
      console.log('found nothing');
    });
  }
});

btSerial.inquire();
```

The code is detecting device properly, but I can't connect to it (Cannot connect error in the console).

Is there something am I missing?

eelcocramer, 2019-02-11:
That error is generated [here](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/54c81109fc90d98b59ddb8d1c0d4e82209960535/src/linux/BTSerialPortBinding.cc#L86) and the reason is that the [connect function](https://linux.die.net/man/3/connect) that [is called here](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/54c81109fc90d98b59ddb8d1c0d4e82209960535/src/linux/BTSerialPortBinding.cc#L69) doet not return successfully.

As it would be useful if the error would be a bit more verbose, I've added the error code to the message and pushed that change into the `next-release` branch. You may want to try that branch and see if you can use the error code to figure out your problem.

max-ci, 2019-02-13:
Thanks @eelcocramer I fixed my issue using GATT in Python.

closed 2019-02-13 by max-ci