eelcocramer/node-bluetooth-serial-port #210: Connection closes automatically

iamsush, 2019-03-29:
If the connection to a bluetooth device is opened again after closing it once. node-bluetooth-serial-port is not able to read the data and it closes the connection automatically.

Getting below error:
```
Bluetooth disconnected
Error: Error reading from connection

Bluetooth disconnected
```

eelcocramer, 2019-03-29:
You should not reuse the connection object but re-create a new one when starting a new connection. Also see #90.

iamsush, 2019-03-29:
By that you mean, for every connection you need to have a new connection object, 
i.e, 
```
var btSerial = new (require('bluetooth-serial-port')).BluetoothSerialPort();
```
?
But isn't this something troublesome, to create the object again and again? Might be possible some enhancements could be there. What do you say?

eelcocramer, 2019-03-29:
It is troublesome :-) I'm happy to consider a PR that fixes this.

iamsush, 2019-03-29:
I could do that, but I am not sure where to start with the code as I have little experience with native code development. On the top layer, I would like to contribute.

eelcocramer, 2019-03-29:
This is the thing. It is a rather complex issue. Re-using the object was never a use case for me personally and other folks seemed to be happy enough with work-around.

iamsush, 2019-03-29:
Actually, the application I am working on has a use case to switch between a serial port and a bluetooth port. So, for me I want to connect and disconnect bluetooth port multiple times. If there had been only one connection object to handle bluetooth port connection, it would have been way better than now for me.

Also, I guess, giving the functionality to connect and reconnect through the same connection object would be useful for a lot of people. (They would not have to worry about deleting the previous object and other memory leak issues.)

_bluetooth-serial-port_  could come out as the universal bluetooth module to use, if working with node js and bluetooth.

(PS: I would still see If I can contribute to the branch, If I got to know how to deal with it. haha)

Thanks man!

eelcocramer, 2019-03-29:
#90 and #162 may provide you with some pointers about the what the problem may be. Reading some of the comments I believe on Windows it is possible to re-use the object so I guess the Linux code is the best place to look for your issue. The implementation can be found in the file [BluetoothSerialPort.cc](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/linux/BluetoothSerialPort.cc).