eelcocramer/node-bluetooth-serial-port #212: ../src/node_buffer.cc:219:char* node::Buffer::Data(v8::Local<v8::Object>): Assertion `obj->IsArrayBufferView()' failed.

PkNuAbrahim, 2019-05-26:
var BTSP = require('bluetooth-serial-port');
var serial = new BTSP.BluetoothSerialPort();

serial.on('found', function(address, name) {

    // you might want to check the found address with the address of your
    // bluetooth enabled Arduino device here.

    if (name === "HC-05"){
    serial.findSerialPortChannel(address, function(channel) {
        serial.connect(address, channel, function() {
            console.log('connected');
            process.stdin.resume();
            process.stdin.setEncoding('utf8');
            console.log('Press "1" or "0" and "ENTER" to turn on or off the light.')
process.stdin.on('data', function (data) {
            
                serial.write(data);
            });

            serial.on('data', function(data) {
                console.log('Received: ' + data);
            });
        }, function () {
            console.log('cannot connect');
        });
    });
}
});

serial.inquire();
i wanna know why this code occur error ( node[4957]: ../src/node_buffer.cc:219:char* node::Buffer::Data(v8::Localv8::Object): Assertion `obj->IsArrayBufferView()' failed.
) i wanna connet arduino and raspberry pi3 with node.js

addaleax, 2019-05-26:
Linking my earlier comment here so that maintainers can see it: https://github.com/nodejs/help/issues/1927#issuecomment-495998642

eelcocramer, 2019-05-27:
@addaleax thanks a lot for the find. Your comment is spot on. The code should return after the throw and the parameter to write must be a buffer.

@PkNuAbrahim if you pass a buffer to `write` you should be find.

I will fix the returns in the next release and keep this issue open as a reference.

closed 2019-05-27 by eelcocramer

eelcocramer, 2019-05-27:
Released 2.2.4 on npm. Does address the return behaviour.

@PkNuAbrahim you should still pass buffer as the parameter and not a string.

PkNuAbrahim, 2019-05-27:
@eelcocramer First, thank you for your release!!
I updated version 2.2.4 and tried again.
And I fixed my code with buffer, but still have another error

process.stdin.on('data', function (data) {
                const buf1 = Buffer.from(data);
                serial.write(buf1);
            });

and this is the error

pi@raspberrypi:~ $ sudo node ys.js
connected
Press "1" or "0" and "ENTER" to turn on or off the light.
1
/home/pi/node_modules/bluetooth-serial-port/lib/bluetooth-serial-port.js:118
            this.connection.write(buffer, this.address, cb);
                            ^

TypeError: Third argument must be a function
    at BluetoothSerialPort.write (/home/pi/node_modules/bluetooth-serial-port/lib/bluetooth-serial-port.js:118:29)
    at ReadStream.<anonymous> (/home/pi/ys.js:18:24)
    at ReadStream.emit (events.js:189:13)
    at addChunk (_stream_readable.js:284:12)
    at readableAddChunk (_stream_readable.js:261:13)
    at ReadStream.Readable.push (_stream_readable.js:220:10)
    at TTY.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)

I'm sorry that I'm not good at coding
Thank you for your kind reply again

eelcocramer, 2019-05-27:
Please check the readme for an example and the [function reference doc](https://github.com/eelcocramer/node-bluetooth-serial-port#bluetoothserialportwritebuffer-callback)

TL;DR; the write function requires an extra parameter that is called by the module after a write did finish.

PkNuAbrahim, 2019-05-29:
@eelcocramer its working good, really appreciate you

eelcocramer, 2019-05-29:
@PkNuAbrahim thank you. Together we are improving open source!

PkNuAbrahim, 2019-05-29:
@eelcocramer your code is working good. but i'm faced with new problem , when i used process.stdin.on('data', function (data) {
var buf1 = Buffer.from(data);
serial.write(buf1,err,bytesWritten);
if(err) console.log(err);
});
this code need console input but i wanna send var value without console input

commit b1bb9739 (2019-05-27): Closes #212