eelcocramer/node-bluetooth-serial-port #215: different Node.js version using NODE_MODULE_VERSION

GavinRoman, 2019-06-05:
### The error is followed:
_A JavaScript error occurred in the main process  A JavaScript error occurred in the main process
node_modules/bluetooth-serial-port/build/Release/BluetoothSerialPort.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 70. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:850:18)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:677:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
    at Function.Module._load (internal/modules/cjs/loader.js:601:3)
    at Module.require (internal/modules/cjs/loader.js:715:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at bindings (/Users/*****/career/app/dongliang/treadmill/node_modules/bindings/bindings.js:76:44)
    at /Users/******/career/app/dongliang/treadmill/node_modules/bluetooth-serial-port/lib/bluetooth-serial-port.js:20:39_

environment: 
os  version : mac os Mojave 10.14.5 (18F132)
node verison: v10.16.0
npm version: 6.9.0

**### The following is the package.json:**
{
  "name": "treadmill",
  "version": "1.0.0",
  "main": "main.js",
  "scripts": {
    "dev": "electron ."
  },
  "license": "MIT",
  "dependencies": {
    "bluetooth-serial-port": "^2.2.4",
    "electron": "^5.0.2"
  },
  "devDependencies": {
    "electron-rebuild": "^1.8.5"
  }
}

eelcocramer, 2019-06-05:
The error message says it all. You build the module against another version of nodejs, after you update node you have to rebuild this module again using ‘npm rebuild’ or ‘npm install’.

closed 2019-06-05 by eelcocramer

GavinRoman, 2019-06-22:
Must I use the appoint node version? Thanks very much

eelcocramer, 2019-06-24:
You can use any of the supported node versions that are listed in the [package.json](package.json) file. But you have to run your program with the same version you build and installed the module with.