eelcocramer/node-bluetooth-serial-port #218: Buffer() deprecation warning

JuanmaLambre, 2019-07-11:
Hi! I just executed the following code with `node` (LTS v10.13.0):

```
var server = new (require('bluetooth-serial-port')).BluetoothSerialPortServer();

var CHANNEL = 1;
var UUID = '38e851bc-7144-44b4-9cd8-80549c6f2912';

server.on('data', function(buffer) {
    console.log('Received data from client: ' + buffer);
});

console.log("Listening for client...")
server.listen(function (clientAddress) {
    console.log('Client: ' + clientAddress + ' connected!');
}, function(error){
  console.error("Something wrong happened!:" + error);
}, {uuid: UUID, channel: CHANNEL} );
```

With an Android app I easily connect via Bluetooth and whenever I send data the output is correct (it prints whatever I send). But I also get the following warning:

```
(node:1079) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
```

Any hint on how this issue can be solved? Thanks in advance!

eelcocramer, 2019-07-11:
Not sure. As far as I can see this module does not use `Buffer()` but always uses `Buffer.from` to create a buffer. Maybe it is used somewhere else in your code?

JuanmaLambre, 2019-07-11:
The code I showed is all the code I have written. Weird enough, now I see the warning is only shown the first time the server receives data. I can only assume the problem must be in other lib. 

Thanks for the answer and this great module :)

eelcocramer, 2019-07-11:
Thanks! +1

Closing this.

closed 2019-07-11 by eelcocramer

jordanorc, 2020-07-31:
I'm having the same problem and as @JuanmaLambre pointed the warning is only shown the first time it receives data, but I'm using as a client:
```
btSerial.findSerialPortChannel("3C:71:BF:FE:71:0E", function(channel) {
    btSerial.connect("3C:71:BF:FE:71:0E", channel, function() {
        console.log('connected');

        btSerial.write(Buffer.from([2, 0x01, 0x01]), function(err, bytesWritten) {
            if (err) console.log(err);
        });

        btSerial.on('data', function(buffer) {
            console.log(buffer);
            console.log("aqui");
        });
    }, function () {
        console.log('cannot connect');
    });
    console.log('closing');
    // close the connection when you're ready
    btSerial.close();
}, function() {
    console.log('found nothing');
});
```

Any idea about what can be happening? Did you find a solution @JuanmaLambre ?