eelcocramer/node-bluetooth-serial-port #219: Building failed

PsychoXIVI, 2019-08-16:
Building issue.

```
node@12.8.1
npm@6.10.2
node-gyp@5.0.3
```

console log: https://pastebin.com/u0duJ1pC

file debug log: https://pastebin.com/pYDgqgRM

tl;dr of log:
```
gyp info spawn args [ 'V=1', 'BUILDTYPE=Release', '-C', 'build' ]
In file included from ../src/linux/BluetoothSerialPort.cc:13:0:
../src/DeviceINQ.h:49:30: error: ‘v8::Handle’ has not been declared
         static void Init(v8::Handle<v8::Object> exports);
                              ^~~~~~
```

Lack of `#include <v8.h>`? Well, `<node.h>` should do it... Idk why there is a problem.

eelcocramer, 2019-08-16:
Currently only stable (LTS) versions of node are supported. You may want to consider using node v10 unless you really need some experimental feature.

PsychoXIVI, 2019-08-16:
And Node 12 is going to be next LTS version.

PsychoXIVI, 2019-08-16:
Found that @ELundby45 already is working on it <3

PsychoXIVI, 2019-08-17:
Okay, it seems to be working. I used @ELundby45 code and added Linux support (which was so similar that I done it nearly without any knowledge or experience about `Nan` and `v8`).

There is a test I used:
```javascript

const { BluetoothSerialPortServer } = require('bluetooth-serial-port')
var server = new BluetoothSerialPortServer();

var CHANNEL = 1;
var UUID = '00001101-0000-1000-8000-00805F9B34FB';

server.on('data', function(buffer) {
    console.log('Received data from client: ' + buffer);

    console.log('Sending data to the client');
    server.write(Buffer.from('it works :) \r\n'), function (err, bytesWritten) {
        if (err) {
            console.log('Error! ' + err.toString());
        } else {
            console.log('Sent ' + bytesWritten + ' to the client!');
        }
    });
});

server.listen(function (clientAddress) {
    console.log('Client: ' + clientAddress + ' connected!');
}, function(error){
	console.error("Something wrong happened!:" + error);
}, {uuid: UUID, channel: CHANNEL} );
```
And it works well with [my testing Bluetooth Flutter mobile app](https://github.com/edufolly/flutter_bluetooth_serial/tree/master/example).

I will test also client in few minutes (just installing git command line on my Windows to make a server).

I am not sure if it would work with Node 10, I don't have time (well, and disk space on my Raspberry Pi I am testing on xD) right now to install and test it.

eelcocramer, 2019-08-18:
Nice work. If it still compiles and works on v10 I would be happy to consider a PR.

eelcocramer, 2019-08-22:
If you are using the server on Linux you may want to try out PR #223. Feedback is highly appreciated.

eelcocramer, 2019-08-22:
Sorry, ignore my last comment I will cherry pick the changes from your repo.

PR #225 (2019-08-23): Releases version 2.2.6
Adds support for node v12.
Removes support for node v6 and v7.

Closes #224 
Closes #219
closed 2019-08-23 by eelcocramer