eelcocramer/node-bluetooth-serial-port #220: Question: Multiple invocations of same function in platform code?

PsychoXIVI, 2019-08-17:
Why there are multiple invocations of the same function in platform code?

In server (Linux) there is only one of such, but these `BTSerialPortBinding` and `DeviceINQ` looks insane... xd

https://github.com/eelcocramer/node-bluetooth-serial-port/blob/1af7838e9c821cdee8a8e4bc6655cb9165b000be/src/linux/BTSerialPortBinding.cc#L236-L241

eelcocramer, 2019-08-17:
The lines seem to be redundant. I wrote this years ago so I’m not sure why I put these in anymore or if they can be left out.

eelcocramer, 2019-08-21:
This was introduced back in the time NaN 2 and node v4 support was added by merging in changes from a forked repository that already added that support into this repository. A quick check makes me believe the lines are redundant and could be removed in a future version.

eelcocramer, 2019-08-23:
These are now removed.

closed 2019-08-23 by eelcocramer