eelcocramer/node-bluetooth-serial-port #227: Unable to use bluetooth-serial-port when trying to build mBlock

MateusRodCosta, 2019-09-11:
Hello.

I'm currently trying to compile [mBlock](https://github.com/Makeblock-official/mBlock) 4.0.0 on Arch Linux with Node 8 but I'm unable to.

According to an user on Matrix, it could be due to an interaction between electron and this library.

Here are the logs I have: https://paste.gnome.org/pa5rpfxcs

npm is using the version 2.2.6 of bluetooth-serial-port

What could be the issue and how should I go about fixing it?

Do note that mBlock codebase is unmaintaned and I have no knowledge about how any of it works.

eelcocramer, 2019-09-11:
Hi, the link to the logs does not work.

MateusRodCosta, 2019-09-11:
Here: https://gist.github.com/MateusRodCosta/75560e0c2113383aa0163a7b120e2a7d

And 

Here: https://paste.gnome.org/ppjmuqphy

eelcocramer, 2019-09-11:
Can you try installing v2.2.5? This may be a regression.

MateusRodCosta, 2019-09-11:
Seems to work: https://gist.github.com/MateusRodCosta/f61c148c517dcf20b6bb415606cee609

eelcocramer, 2019-09-11:
Ok, thanks for trying. It seems that in your setup the module is not building while the automated build for node 8 on linux works fine. I will need to figure out why this is the can. May be arm related?

Can you show the output of ‘npm ls’

eelcocramer, 2019-09-13:
I've tried building the module for node v8 on both `armv7` and `arm64v8` and everything seems to be building fine. I not able to test on arch linux. I'm closing this for now as I'm not convinced it is related to this module.

closed 2019-09-13 by eelcocramer