eelcocramer/node-bluetooth-serial-port #229: Test build Linux using docker fails

gmacario, 2020-01-07:
Hello,

I followed the instructions at section "Test build Linux using docker" of README.md 

```bash
docker build -t bluetooth-serial-port .
```

but was unable to complete step "RUN npm install --unsafe-perm":

```
gpmacario@nemo:~/github/gmacario/node-bluetooth-serial-port (master)$ docker build -t bluetooth-serial-port .                                                                Sending build context to Docker daemon  984.6kB                                                                                                                              Step 1/6 : FROM node:latest                                                                                                                                                   ---> 7354979df4ec                                                                                                                                                           Step 2/6 : RUN apt-get update                                                                                                                                                 ---> Using cache                                                                                                                                                             ---> 49f6abc5cb12                                                                                                                                                           Step 3/6 : RUN apt-get install -y build-essential libbluetooth-dev                                                                                                            ---> Using cache                                                                                                                                                             ---> 033333f4e474                                                                                                                                                           Step 4/6 : ADD . node-bluetooth-serial-port                                                                                                                                   ---> Using cache                                                                                                                                                             ---> 7d68f99b6a69                                                                                                                                                           Step 5/6 : WORKDIR node-bluetooth-serial-port                                                                                                                                 ---> Using cache                                                                                                                                                             ---> d9a0f0debb57                                                                                                                                                           Step 6/6 : RUN npm install --unsafe-perm                                                                                                                                      ---> Running in 292c6b5f71bc                                                                                                                                                                                                                                                                                                                             > bluetooth-serial-port@2.2.6 install /node-bluetooth-serial-port                                                                                                            > node-gyp configure build                                                                                                                                                                                                                                                                                                                                make: Entering directory '/node-bluetooth-serial-port/build'                                                                                                                   CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/BluetoothSerialPort.o                                                                                           CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o                                                                                                   ../src/linux/DeviceINQ.cc: In static member function 'static void DeviceINQ::EIO_AfterSdpSearch(uv_work_t*)':                                                                ../src/linux/DeviceINQ.cc:139:28: warning: 'v8::Local Nan::Callback::Call(int, v8::Local*) const' is deprecated [-Wdeprecated-declarations]                 baton->cb->Call(1, argv);                                                                                                                                                                           ^                                                                                                                                                In file included from ../src/linux/DeviceINQ.cc:14:0:                                                                                                                        ../node_modules/nan/nan.h:1740:3: note: declared here                                                                                                                           Call(int argc, v8::Local argv[]) const {                                                                                                                          ^~~~                                                                                                                                                                      ../src/linux/DeviceINQ.cc: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE DeviceINQ::InquireSync(Nan::NAN_METHOD_ARGS_TYPE)':                                 ../src/linux/DeviceINQ.cc:251:26: warning: 'v8::Local Nan::Callback::Call(int, v8::Local*) const' is deprecated [-Wdeprecated-declarations]                   found->Call(2, argv);                                                                                                                                                                           ^                                                                                                                                                  In file included from ../src/linux/DeviceINQ.cc:14:0:                                                                                                                        ../node_modules/nan/nan.h:1740:3: note: declared here                                                                                                                           Call(int argc, v8::Local argv[]) const {                                                                                                                          ^~~~                                                                                                                                                                      ../src/linux/DeviceINQ.cc:255:27: warning: 'v8::Local Nan::Callback::Call(int, v8::Local*) const' is deprecated [-Wdeprecated-declarations]                 callback->Call(0, argv);                                                                                                                                                                           ^                                                                                                                                                 In file included from ../src/linux/DeviceINQ.cc:14:0:                                                                                                                        ../node_modules/nan/nan.h:1740:3: note: declared here                                                                                                                           Call(int argc, v8::Local argv[]) const {                                                                                                                          ^~~~                                                                                                                                                                      ../src/linux/DeviceINQ.cc: In member function 'virtual void InquireWorker::Han [... cut ...]                                                                                            The command '/bin/sh -c npm install --unsafe-perm' returned a non-zero code: 1                                                                                               gpmacario@nemo:~/github/gmacario/node-bluetooth-serial-port (master)$
```

Reproduced on

- Ubuntu 18.04.3 LTS 64-bit
- Docker version 19.03.5, build 633a0ea838
- latest commit on master branch (89e62bc5fb1a6763fd3711505d4583efb4bd5449)

eelcocramer, 2020-01-07:
Thank you for the report.

I can reproduce this. The quick fix is to change the base image to `node:lts`. I will push a fix that has this change soon.

closed 2020-01-07 by eelcocramer

gmacario, 2020-01-08:
Hello @eelcocramer,
Thanks for your super-fast response!
II confirm that building against node:LTS (12.14.1 as of today) works fine and produces only warnings during "npm install". 

Just wondering whether either you or someone else has any plans for making this component install on Node 13...

eelcocramer, 2020-01-08:
Currently I only support the LTS versions of node. In the past I’ve found the current versions to be to unstable to work with.

PR #230 (2020-01-07): Release version 2.2.7
Fixes #229
commit 6e8f423d (2020-01-07): Release version 2.2.7 (#230)

* Fixes issue with bulding a docker container
* Releases version 2.2.7

Closes #229