eelcocramer/node-bluetooth-serial-port #231: Windows 10 build still breaking

gnesher, 2020-01-13:
Using Node v12.14.1 (LTS) and Windows 10 I get the following errors during install

'''
..\src\windows\DeviceINQ.cc(180): warning C4473: 'sscanf_s' : not enough arguments passed for format string (compiling source file ..\src\windows\DeviceINQ.cc) [C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
  ..\src\windows\DeviceINQ.cc(180): note: placeholders and their parameters expect 2 variadic arguments, but 1 were provided (compiling source file ..\src\windows\DeviceINQ.cc)
  ..\src\windows\DeviceINQ.cc(180): note: the missing variadic argument 2 is required by format string '%18[^)]' (compiling source file ..\src\windows\DeviceINQ.cc)
  ..\src\windows\DeviceINQ.cc(180): note: this argument is used as a buffer size (compiling source file ..\src\windows\DeviceINQ.cc)
'''

This should have been solved with https://github.com/eelcocramer/node-bluetooth-serial-port/issues/182 but the errors seem identical and the installation fails.

P.S
There's also a large number of warnings, but I assume this is unrelated.

eelcocramer, 2020-01-13:
Thanks for reporting. Can you post the full log of npm? What version of the module are you using?

gnesher, 2020-01-14:
@eelcocramer sure.

Ok, this might be limited to `electron-rebuild` - I do get a very similar output when running npm install (which confused me) but it seems the errors are being reported as warning (and the build does pass). 

I've posted both logs here (verbose) - hopes this helps. I'm using the latest version of bluetooth-serial-port (2.2.7)

This is the error I get when running electron-rebuild with node-bluetooth-serial-port:

```
$ ./node_modules/.bin/electron-rebuild
- Searching dependency tree
× Rebuild Failed

An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@6.1.0
gyp info using node@12.14.1 | win32 | x64
gyp info find Python using Python version 2.7.17 found at "C:\Python27\python.exe"
gyp info find VS using VS2015 (14.0) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio 14.0"
gyp info find VS run with --verbose for detailed information
gyp info spawn C:\Python27\python.exe
gyp info spawn args [
gyp info spawn args   'C:\\Users\\gnesher\\work\\nortec\\electron-tests\\serial-bluetooth-port\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\gnesher\\work\\nortec\\electron-tests\\serial-bluetooth-port\\node_modules\\bluetooth-serial-port\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\gnesher\\work\\nortec\\electron-tests\\serial-bluetooth-port\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\gnesher\\.electron-gyp\\7.1.7\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\gnesher\\.electron-gyp\\7.1.7',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\gnesher\\work\\nortec\\electron-tests\\serial-bluetooth-port\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\gnesher\\\\.electron-gyp\\\\7.1.7\\\\<(target_arch)\\\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\gnesher\\work\\nortec\\electron-tests\\serial-bluetooth-port\\node_modules\\bluetooth-serial-port',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Users\\gnesher\\work\\nortec\\electron-tests\\serial-bluetooth-port\\node_modules\\bluetooth-serial-port\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64'
gyp info spawn args ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  BluetoothSerialPort.cc
  DeviceINQ.cc
  BTSerialPortBinding.cc
  BluetoothHelpers.cc
  win_delay_load_hook.cc
c:\users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(8515): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\windows\BluetoothSerialPort.cc) [C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
  c:\users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(6919): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\windows\BluetoothSerialPort.cc)
c:\users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(8524): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\windows\BluetoothSerialPort.cc) [C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
  c:\users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(6919): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\windows\BluetoothSerialPort.cc)
C:\Users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(8515): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\windows\DeviceINQ.cc) [C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
  C:\Users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(6919): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\windows\DeviceINQ.cc)
C:\Users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(8524): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\windows\DeviceINQ.cc) [C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
  C:\Users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(6919): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\windows\DeviceINQ.cc)
C:\Users\gnesher\.electron-gyp\7.1.7\include\node\v8.h(8515): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\windows\BTSerialPortBinding.cc) [C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\bluetoot [... cut ...]
    at SafeSubscriber._next (C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\spawn-rx\lib\src\index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.next (C:\Users\gnesher\work\nortec\electron-tests\serial-bluetooth-port\node_modules\rxjs\internal\Subscriber.js:143:22)

gnesher@DESKTOP-BTD711A MINGW64 ~/work/nortec/electron-tests/serial-bluetooth-port (master)
```

eelcocramer, 2020-01-14:
Weird. The automatic CI build for windows passes without problems so I'm unsure why you are getting this issue.

I don't have a windows development environment so I'm not able to check this with electron.

Raffone17, 2020-01-15:
We had the same problem with electron 7, the problem is that the version of the javascript v8 engine it uses when compiling is higher than that used by node 12.4.1, so a lot of deprecated functions no longer exist. Try using electron 6.

closed 2020-02-06 by eelcocramer