eelcocramer/node-bluetooth-serial-port #232: unable to get the RFCOMM  channel on myBtPort.findSerialPortChannel()

sujitpatil72, 2020-01-27:
Hi, I am unable to get the RFCOMM  channel after passing the address to  myBtPort.findSerialPortChannel().

I am getting callback on 'found' event as follows.

Found: (DC:A2:66:28:6D:E2) with name
Found: (AD:AA:6D:DE:6D:8C)PEBBLE XS with name PEBBLE XS
Found: (94:65:2D:E9:B5:28)OnePlus 5T with name OnePlus 5T
Found: (F8:77:B8:0E:F0:3F)[TV] UA55JU6470 with name [TV] UA55JU6470
scan did finish.

So as seen above scanning did finish without consoling the RFCOMM  channel. Please help me in solving this.

Thanks in advance :)

My Node version: 10.16.0
My code as below:

`var BluetoothSerialPort = require("../lib/bluetooth-serial-port.js").BluetoothSerialPort;
var myBtPort = new BluetoothSerialPort();
var dataBuffer = "";

myBtPort.on('found', function (address, name) {
  console.log('Found: ' + address + ' with name ' + name);

  myBtPort.findSerialPortChannel(address, function (channel) {
    console.log('Found RFCOMM channel for serial port on ' + name + ': ' + channel);

    console.log('Attempting to connect...');

    myBtPort.connect(address, channel, function () {
      console.log('Connected. Receiving data...');

      myBtPort.on('data', function (buffer) {

        dataBuffer = dataBuffer + buffer.toString('utf-8');

        if (dataBuffer.indexOf("\n") != -1) {
          console.log("New incoming string");
          console.log("String = " + dataBuffer);
          dataBuffer = "";
        }

      });
    });
  });
});

myBtPort.on('finished', function () {
  console.log('scan did finish');
});

myBtPort.inquire();`

eelcocramer, 2020-01-27:
Maybe your devices do not support the RFCOMM profile.

sujitpatil72, 2020-01-27:
Ok, I will check that.
Thanks for quick reply @eelcocramer.

closed 2020-01-27 by sujitpatil72