eelcocramer/node-bluetooth-serial-port #233: "Error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments" while installing

Sv443, 2020-02-28:
I'm trying to install this package with `npm i bluetooth-serial-port` and after switching from Python 3 to 2.7 (since it threw another error) it has been throwing me this new error a few times while installing:  
```Error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments```  
This error is happening while node-gyp is building the BluetoothSerialPort project.  
There's also a bunch of warnings.  
[Here's the full error log](https://pastebin.com/bqUG9JjV).  
  
OS: Windows 10 (1909 - Build 18363.657)  
Node.js version: 13.9.0  
NPM version: 6.13.7  
Python version: 2.7.16

eelcocramer, 2020-02-29:
Node v13 is currently not supported please use v12 with this module.

Sv443, 2020-03-05:
> 
> 
> Node v13 is currently not supported please use v12 with this module.

Okay well I'm glad there's at least a temporary workaround to it. Although with the new V8 update coming to Node soon I'd want to be able to switch to v13 entirely, not having some projects require NVM. Would it be possible to update the package to support Node 13 in the near future?

eelcocramer, 2020-03-05:
It is possible to update the package and add node 13 support but I do not have plans to add this myself.

I'm happy to accept a PR that adds support though.

jithinjosejacob, 2020-11-12:
I faced the same issue today while using Node v13 in Azure Pipelines, npm install failed with same error.I had to used Node version 12.16.1 to resolve this

1pavanb, 2020-11-16:
What the hell is this??
since 12.16.1 is not LTS I used 10.23.0
I tried windows build tools 2019, 2017, 2015
node 14.15.1(LTS), 12.19.1(LTS)

nothing worked. When changed to 10, started working

dhc-zhuchuanpeng, 2020-12-07:
v12.18.3(LTS)

+1

talentdev2020, 2021-09-03:
So does this only work on v12 or 10?

eelcocramer, 2021-09-03:
See #309

talentdev2020, 2021-09-03:
Thanks, @eelcocramer,

Then How can I fix this issue?
[image]

eelcocramer, 2021-09-03:
You may try the fix in the branch mentioned in #309. It would be great if you can provide feedback wether that works or not. CI suggest that the fix in that branch builds on windows.

talentdev2020, 2021-09-03:
Sorry, I can't see any solutions there

talentdev2020, 2021-09-03:
Did you mean these?
- Needs Visual Studio (Visual C++) and its command line tools installed.
- Needs Python 2.x installed and accessible from the command line path.
I installed Python 3.8, should I remove it?

talentdev2020, 2021-09-03:
I tried to install Visual C++ and it's command line tools, but same issue

eelcocramer, 2021-09-04:
I meant you could try out the fix in the branch. 

> Working on a fix, please check the `next-release` branch and let me know if it works (or does not work) for you!

talentdev2020, 2021-09-05:
Why does this need for my work?

eelcocramer, 2021-09-05:
I'm not sure I understand your question.

The current version of the module does not compile against newer versions of nodejs. The patch in the branch does compile. I'm just not sure wether it is functionally ok so I'm reaching out to the community to validate that branch.

talentdev2020, 2021-09-05:
In that branch, you are going to install bluetooth-serial-port library, but I am not using it.
The gyp error is a common issue in Node 14 and 16. If I use Node 12, it is working.
So my question is about how we can fix gyp error on Node 14 and 16

eelcocramer, 2021-09-05:
If you think its an issue with gyp you check with them. I'm sure they will be very supportive...

business-helper, 2021-12-04:
> I faced the same issue today while using Node v13 in Azure Pipelines, npm install failed with same error.I had to used Node version 12.16.1 to resolve this

Thanks, it worked for me!