eelcocramer/node-bluetooth-serial-port #234: NODE_MODULE_VERSION error

lrosique, 2020-03-03:
Hi,

I'm trying to use node-bluetooth-serial-port with ionic + electron on windows 10 but I have this error :

main-es2015.a710b622e5cb8255cd22.js:1 ERROR Error: Uncaught (in promise): Error: The module '\\?\C:\workspaces\ionic\ts_dashboard\ts_dashboard\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 76. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Here's my environment :
- Windows 10
- bluetooth-serial-port : ^2.2.7
- angular : ~8.1.2
- ionic : ^5.0.0
- electron : ^8.0.0
- Node : v12.14.1
- npm : 6.13.4

Can you help me please ? I don't know which version of Node is compatible with the library.

eelcocramer, 2020-03-03:
You need to rebuild this module because it was previously build with another node version.

lrosique, 2020-03-03:
thank you for your response, which node version did you use to build it previously ?
i tried to rebuild the module but it crashes everytime

gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\workspaces\ionic\ts_dashboard\ts_dashboard\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:223:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\workspaces\\ionic\\ts_dashboard\\ts_dashboard\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=8.0.3" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd C:\workspaces\ionic\ts_dashboard\ts_dashboard\node_modules\bluetooth-serial-port
gyp ERR! node -v v12.14.1
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok

Failed with exit code: 1

eelcocramer, 2020-03-03:
You need to rebuild. There is no binary in the package it is build locally on ‘npm install’. You may have to reinstall the module.

eelcocramer, 2020-03-04:
The command is probably: `npm rebuild <module name>`

closed 2020-03-04 by eelcocramer