eelcocramer/node-bluetooth-serial-port #235: could not run server

SoGoDev, 2020-06-24:
I have such code , and when i check if server is running, it always return false.

import * as btSerial from "bluetooth-serial-port";

export default class Bluetooth {
  private bl;
  CHANNEL = 10; // My service channel. Defaults to 1 if omitted.
  UUID = '38e851bc-7144-44b4-9cd8-80549c6f2912'; // My own service UUID. Defaults to '1101' if omitted
  constructor() {
    this.onData = this.onData.bind(this);
    this.onError = this.onError.bind(this);
    this.onSuccess = this.onSuccess.bind(this);

    this.bl = new btSerial.BluetoothSerialPortServer();

    this.bl.listen(this.onSuccess, this.onError);
    this.bl.on('data', this.onData, {channel: this.CHANNEL})

  }

  onSuccess(address) {
    console.log('Client: ' + address + ' connected!');
  }

  onError(error) {
    console.error("Something wrong happened!:" + error);
  }

  onData(buffer){
    console.log('Received data from client: ' + buffer);
  }

}

zxfrank, 2020-08-14:
same for me, raspberryPI, it must be a bluetooth config issue...

zxfrank, 2020-08-14:
edit 
```
/etc/systemd/system/dbus-org.bluez.service
```

change the line(s):
```
ExecStart=/usr/lib/bluetooth/bluetoothd -C --noplugin=sap
ExecStartPost=/usr/bin/sdptool add SP
```

reboot, then `sudo sdptool browse local`
should output something like:

```
[...]
Service Name: Serial Port
Service Description: COM Port
Service Provider: BlueZ
Service RecHandle: 0x10003
Service Class ID List:
  "Serial Port" (0x1101)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 1
Language Base Attr List:
  code_ISO639: 0x656e
  encoding:    0x6a
  base_offset: 0x100
Profile Descriptor List:
  "Serial Port" (0x1101)
    Version: 0x0100
[...]
```

zxfrank, 2020-08-14:
Tried with https://github.com/dinohorvat/pybluez--rfcomm-server
```
sudo python rfcomm-server.py 
Waiting for connection on RFCOMM channel 1
```
`sudo sdptool browse local` can see it, but can't connect from android bluetooth serial
```
[...]
Service Name: SampleServer
Service RecHandle: 0x10007
Service Class ID List:
  UUID 128: 94f39d29-7d6d-437d-973b-fba39e49d4ee
  "Serial Port" (0x1101)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 1
Profile Descriptor List:
  "Serial Port" (0x1101)
    Version: 0x0100
[...]
```