eelcocramer/node-bluetooth-serial-port #236: Command failed at node-gyp configure build

kaitoAckerman, 2020-06-26:
Running under MacOs Catalina. 

```
▶ yarn add bluetooth-serial-port              
yarn add v1.22.4
[1/4] 🔍  Resolving packages...
[2/4] 🚚  Fetching packages...
[3/4] 🔗  Linking dependencies...
[4/4] 🔨  Building fresh packages...
[1/3] ⠂ core-js
[2/3] ⠂ bluetooth-serial-port
error /Users/kaitoackerman/Leaf/DesktopApp/ThermoScanner/node_modules/bluetooth-serial-port: Command failed.
Exit code: 127
Command: node-gyp configure build
Arguments: 
Directory: /Users/kaitoackerman/Leaf/DesktopApp/ThermoScanner/node_modules/bluetooth-serial-port
Output:

```

need help @eelcocramer

kaitoAckerman, 2020-06-26:
Same result when using `npm install`

eelcocramer, 2020-06-26:
You are leaving out the relevant section of the log (where the error happens). That said you may want to check if the version of node you are using is supported by the module. Also I do not maintain the module anymore unless there are any critical bugs that need to fixed.

If you go through the forks of this project you may find a fork that supports more recent versions of node.

closed 2020-07-02 by eelcocramer