eelcocramer/node-bluetooth-serial-port #259: Failed to install

eyalmnm, 2020-10-04:
Hi,
I am trying to install the library but I got the following errors:
`Eyals-MacBook-Pro-2:Electron_Base eyalmuchtar$ npm install bluetooth-serial-port
 

> bluetooth-serial-port@2.2.7 install /Users/eyalmuchtar/Projects/Moblers/Electron_Base/node_modules/bluetooth-serial-port
> node-gyp configure build

  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o
../src/osx/DeviceINQ.mm:75:16: warning: 'Call' is deprecated
      [-Wdeprecated-declarations]
        found->Call(2, argv);
               ^
../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:78:15: warning: 'Call' is deprecated
      [-Wdeprecated-declarations]
    callback->Call(0, 0);
              ^
../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:109:16: warning: 'Call' is deprecated
      [-Wdeprecated-declarations]
    baton->cb->Call(1, argv);
               ^
../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:136:5: warning: ignoring return value of function
      declared with 'warn_unused_result' attribute [-Wunused-result]
    target->Set(ctx, Nan::New("DeviceINQ").ToLocalChecked(), t->GetFunction(ctx).ToLocalChecked());
    ^~~~~~~~~~~ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/osx/DeviceINQ.mm:227:16: warning: 'Call' is deprecated
      [-Wdeprecated-declarations]
        found->Call(2, argv);
               ^
../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:230:15: warning: 'Call' is deprecated
      [-Wdeprecated-declarations]
    callback->Call(0, 0);
              ^
../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:296:20: error: no matching member function for call to
      'Set'
        deviceObj->Set(Nan::New("name").ToLocalChecked(), Nan::New([devi...
        ~~~~~~~~~~~^~~
/Users/eyalmuchtar/Library/Caches/node-gyp/14.4.0/include/node/v8.h:3639:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
[... 99 lines cut ...]
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Darwin 17.7.0
gyp ERR! command "/usr/local/Cellar/node/14.4.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build"
gyp ERR! cwd /Users/eyalmuchtar/Projects/Moblers/Electron_Base/node_modules/bluetooth-serial-port
gyp ERR! node -v v14.4.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN electorn-bluetooth@1.0.0 No repository field.
npm WARN electorn-bluetooth@1.0.0 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-serial-port@2.2.7 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-serial-port@2.2.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/eyalmuchtar/.npm/_logs/2020-10-04T16_45_54_599Z-debug.log
Eyals-MacBook-Pro-2:Electron_Base eyalmuchtar$ 
Eyals-MacBook-Pro-2:Electron_Base eyalmuchtar$ node-gyp configure buil
-bash: node-gyp: command not found
Eyals-MacBook-Pro-2:Electron_Base eyalmuchtar$ 
`

eelcocramer, 2020-10-04:
Hi. Thanks for your intrest is using the module. I do not maintain this anymore but there is a fork that is active and seems to have move forward on some of the issues this module currently has. You may want to check it out: https://github.com/tinyprinter/node-bluetooth-serial-port

tsapko3628, 2021-02-09:
Hi. If I send you a pull request, will you merge it?

eelcocramer, 2021-02-09:
Thanks for reaching out.

I will definitely consider any PR. Of course I need to do some checks before merging it so I cannot make any promises.