eelcocramer/node-bluetooth-serial-port #309: Won't install with current LTS node 14

peter9091, 2021-06-03:
I have installed node 14 (lts), which works fine, i could install nemerous modules without any problems.
Now trying installing this module i've got an error:

```
root@localhost:~/node$ sudo npm install bluetooth-serial-port

> bluetooth-serial-port@2.2.7 install /root/node/node_modules/bluetooth-serial-port
> node-gyp configure build

make: Entering directory '/root/node/node_modules/bluetooth-serial-port/build'
  CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/BluetoothSerialPort.o
In file included from ../src/linux/BluetoothSerialPort.cc:12:
/root/.cache/node-gyp/14.17.0/include/node/node.h:758:43: warning: cast between incompatible function types from ‘void (*)(v8::Local<v8::Object>)’ to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
       (node::addon_register_func) (regfunc),                          \
                                           ^
/root/.cache/node-gyp/14.17.0/include/node/node.h:792:3: note: in expansion of macro ‘NODE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
   ^~~~~~~~~~~~~
../src/linux/BluetoothSerialPort.cc:23:1: note: in expansion of macro ‘NODE_MODULE’
 NODE_MODULE(BluetoothSerialPort, InitAll)
 ^~~~~~~~~~~
  CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o
../src/linux/DeviceINQ.cc: In static member function ‘static void DeviceINQ::EIO_AfterSdpSearch(uv_work_t*)’:
../src/linux/DeviceINQ.cc:139:28: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
     baton->cb->Call(1, argv);
                            ^
In file included from ../src/linux/DeviceINQ.cc:14:
../../nan/nan.h:1742:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/linux/DeviceINQ.cc: In static member function ‘static void DeviceINQ::Init(v8::Local<v8::Object>)’:
../src/linux/DeviceINQ.cc:166:98: warning: ignoring return value of ‘v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)’, declared with attribute warn_unused_result [-Wunused-result]
     target->Set(ctx, Nan::New("DeviceINQ").ToLocalChecked(), t->GetFunction(ctx).ToLocalChecked());
                                                                                                  ^
In file included from ../src/linux/DeviceINQ.cc:12:
/root/.cache/node-gyp/14.17.0/include/node/v8.h:3670:37: note: declared here
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
../src/linux/DeviceINQ.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE DeviceINQ::InquireSync(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/linux/DeviceINQ.cc:251:26: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
       found->Call(2, argv);
                          ^
In file included from ../src/linux/DeviceINQ.cc:14:
../../nan/nan.h:1742:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/linux/DeviceINQ.cc:255:27: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
     callback->Call(0, argv);
                           ^
In file included from ../src/linux/DeviceINQ.cc:14:
../../nan/nan.h:1742:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/linux/DeviceINQ.cc: In member function ‘virtual void InquireWorker::HandleOKCallback()’:
../src/linux/DeviceINQ.cc:284:26: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
       found->Call(2, argv);
                          ^
In file included from ../src/linux/DeviceINQ.cc:14:
../../nan/nan.h:1742:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/linux/DeviceINQ.cc:288:27: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
     callback->Call(0, argv);
                           ^
In file included from ../src/linux/DeviceINQ.cc:14:
../../nan/nan.h:1742:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/linux/DeviceINQ.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE DeviceINQ::SdpSearch(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/linux/DeviceINQ.cc:335:88: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
[... 347 lines cut ...]
182 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
182 verbose stack     at ChildProcess.emit (events.js:376:20)
182 verbose stack     at maybeClose (internal/child_process.js:1055:16)
182 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
183 verbose pkgid bluetooth-serial-port@2.2.7
184 verbose cwd /root/node
185 verbose Linux 5.10.14+
186 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "bluetooth-serial-port"
187 verbose node v14.17.0
188 verbose npm  v6.14.13
189 error code ELIFECYCLE
190 error errno 1
191 error bluetooth-serial-port@2.2.7 install: `node-gyp configure build`
191 error Exit status 1
192 error Failed at the bluetooth-serial-port@2.2.7 install script.
192 error This is probably not a problem with npm. There is likely additional logging output above.
193 verbose exit [ 1, true ]
```

System is a Debian 10 on 32bit ARM (armv7l) A20 from Allwinner.

hope for fix :-)

with best regards,
Peter

eelcocramer, 2021-06-04:
Hi Peter.

Thanks for your report. I deprecated this module some time ago after maintaining it for quite some years. You may want to check out the forks of this module as some of them may have fixes for this issue.

Good luck!

eelcocramer, 2021-08-06:
Working on a fix, please check the `next-release` branch and let me know if it works (or does not work) for you!

eelcocramer, 2021-08-07:
On macos there is an issue. I believe the async behavior using the background worker does not work anymore on the latest sdk.

talentdev2020, 2021-09-03:
Sorry, I can't see any solutions there