eelcocramer/node-bluetooth-serial-port #344: I encountered a fatal error using node-bluetooth-serial-port

Liting1, 2021-09-05:
node: 12.20.2
node-gyp: 8.2.0

for (const device of devices) {
                           ^

TypeError: devices is not iterable
    at BluetoothSerialPort.finish (E:\zkt\doc\test\cmd\node_modules\node-bluetooth-serial-port\lib\bluetooth-serial-port.js:36:28)
[Finished in 6.8s with exit code 1]

Liting1, 2021-09-05:
I took care of it myself

Here's the problem，The correct answer would be the following

[image]

Too bad you wrote it like this

[image]

Liting1, 2021-09-05:
@eelcocramer

eelcocramer, 2021-09-05:
I cannot find the code you quoting in the repo. Can you link to the line?

Liting1, 2021-09-05:
@eelcocramer  This is line 35

https://github.com/tinyprinter/node-bluetooth-serial-port/blob/master/lib/bluetooth-serial-port.js

Liting1, 2021-09-05:
I am looking at the wrong place

Liting1, 2021-09-05:
@Tinyprinter

closed 2021-09-05 by eelcocramer