eelcocramer/node-bluetooth-serial-port #368: Have problem running example code

semilt, 2021-12-09:
I just managed to successfully installed this package, and when I tried to run the Basic client usage I found an error like this

> D:\Coding\Project\win_tablet_test\node_modules\node-bluetooth-serial-port\lib\bluetooth-serial-port.js:37
      for (const device of devices) {
                           ^

> TypeError: devices is not iterable
    at BluetoothSerialPort.finish (D:\Coding\Project\win_tablet_test\node_modules\node-bluetooth-serial-port\lib\bluetooth-serial-port.js:37:28)

I'm using
windows 10
node v14.16.1
npm v6.14.12

can anyone help me?

semilt, 2021-12-09:
Oh, sorry about that :(

closed 2021-12-09 by semilt

eelcocramer, 2021-12-09:
I'm not quite sure how this issue and @hello-smile6's comment end up here and how this is a problem in this module? But the quoted  code

```
D:\Coding\Project\win_tablet_test\node_modules\node-bluetooth-serial-port\lib\bluetooth-serial-port.js:37
for (const device of devices) {
```

is not part of this repository, see [line 37 of that file right here](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/5f8228210a0c1f97080ad3f4b160918d679c2980/lib/bluetooth-serial-port.js#L37).

Then again, the issue is already closed so no worries.

Thanks.