eelcocramer/node-bluetooth-serial-port #378: Problem with installation

thefakhreddin, 2022-01-08:
Hello.
I've been trying to install this package for days via `npm i` under the title of **bluetooth-serial-port** which was not successful.
After going through days of struggle and almost checking every possible thing, I accidentally noticed that that the name of the repo is in fact **node-bluetooth-serial-port** but the readme file instructed to use **npm i bluetooth-serial-port** (the word "node" is missing).
And I guess that was it. A simple name confusion.
If I'm correct, could you please clarify this in the readme to prevent further confusion?
Thanks.

eelcocramer, 2022-01-08:
Thank you for your comment. 

The name this module is registered with on npm is bluetooth-serial-port. Hence the instruction. 

https://www.npmjs.com/package/bluetooth-serial-port

eelcocramer, 2022-01-08:
As a matter of fact you are installing a fork of this module. 

https://www.npmjs.com/package/node-bluetooth-serial-port

The repo for that cab be found here:

https://github.com/tinyprinter/node-bluetooth-serial-port

thefakhreddin, 2022-01-08:
@eelcocramer 

Thanks for your quick response.
So I ran `npm i bluetooth-serial-port --save` again. I keep getting this long error. Do you have any idea what the problem can be? I've been struggling with it for days now.

node -v => 16.13.1
npm -v => 8.3.0
My node-gyp is at 8.4.1 and it's set to use python 3.9 and msvs 2017. I've also tried msvs 2019 [node-gyp instructions](https://www.npmjs.com/package/node-gyp).

I'm wondering what's the right **versioning combination** of the above?

Here's my error:
```
PS G:\Fakhreddin A\Desktop\myProject> npm i bluetooth-serial-port --save
npm ERR! code 1
npm ERR! path G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c node-gyp configure build
npm ERR! Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
npm ERR!   BluetoothSerialPort.cc
npm ERR!   DeviceINQ.cc
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(44): warning C4996: '_WSAQuerySetA::dwSize': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(1320): note: see declaration of '_WSAQuerySetA::dwSize'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(45): warning C4996: '_WSAQuerySetA::dwNameSpace': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(1325): note: see declaration of '_WSAQuerySetA::dwNameSpace'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(46): warning C4996: '_WSAQuerySetA::lpServiceClassId': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(1322): note: see declaration of '_WSAQuerySetA::lpServiceClassId'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(47): warning C4996: '_WSAQuerySetA::dwNumberOfCsAddrs': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(1331): note: see declaration of '_WSAQuerySetA::dwNumberOfCsAddrs'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(48): warning C4996: '_WSAQuerySetA::lpszContext': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(1327): note: see declaration of '_WSAQuerySetA::lpszContext'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(61): warning C4996: '_WSAQuerySetA::lpcsaBuffer': Use WSAQUERYSETW instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(1332): note: see declaration of '_WSAQuerySetA::lpcsaBuffer'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(52): warning C4996: 'WSALookupServiceBeginA': Use WSALookupServiceBeginW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(3628): note: see declaration of 'WSALookupServiceBeginA'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(58): warning C4996: 'WSALookupServiceNextA': Use WSALookupServiceNextW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winsock2.h(3679): note: see declaration of 'WSALookupServiceNextA'
npm ERR! g:\fakhreddin a\desktop\myproject\node_modules\bluetooth-serial-port\src\windows\deviceinq.cc(104): warning C4996: 'Nan::Callback::Call': was declared deprecated [G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
npm ERR!   g:\fakhreddin a\desktop\myproject\node_modules\nan\nan.h(1744): note: see declaration of 'Nan::Callback::Call'
[... cut ...]
npm ERR! gyp ERR! command "G:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Fakhreddin A\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "build"
npm ERR! gyp ERR! cwd G:\Fakhreddin A\Desktop\myProject\node_modules\bluetooth-serial-port
npm ERR! gyp ERR! node -v v16.13.1
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Fakhreddin A\AppData\Local\npm-cache\_logs\2022-01-08T12_09_10_078Z-debug-0.log
PS G:\Fakhreddin A\Desktop\myProject>
```

eelcocramer, 2022-01-08:
See #309

tl;dr; this module is deprecated and does not support node >= 1.13. Fix may be available in `next-release` branch.

thefakhreddin, 2022-01-08:
@eelcocramer Thank you. I'm trying the branch. 
I'm getting this error now:
```
node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing
```
I tried [this](https://stackoverflow.com/questions/46384591/node-was-compiled-against-a-different-node-js-version-using-node-module-versio) but it didn't work out.

I have never had this error before, to what app is the version referring to (93/89)? I guess I need to downgrade a few things.
Again, what is the latest version of node that supports your module? I probably need to downgrade my node.

eelcocramer, 2022-01-08:
This happens when you install the module with one version of node en run it with another version. You should stick with one version for building and running. 

The master branch (and npm stored code) should work with node 12.

thefakhreddin, 2022-01-08:
@eelcocramer 
Thanks for your hint. But I have no idea how to fix it.
I couldn't find anything useful online either. Could you please tell me how to sync these versions?

eelcocramer, 2022-01-08:
Just remove and re-install.

thefakhreddin, 2022-01-08:
@eelcocramer 
I've done that already.
I deleted the whole node_modules folder and ran `npm I`. Still no luck!

eelcocramer, 2022-01-08:
That should be enough when you ran `npm i bluetooth-serial-port` but when you build from source (so run `npm install` from the root of the checkout version of this repo) you should remove the build artefacts from the `build` folder or run `npm rebuild`.