eelcocramer/node-bluetooth-serial-port #413: Segmentation fault when running code on nodejs webserver

valdemartti, 2022-07-03:
Hey! I have a problem, where the package gives segmentation fault when trying to use it. I'm on node v.16.13.0 and npm v.8.1.0, the issue persisted throughout the latest npm installation, the git master branch (3.0.0 I believe) and the next-release branch. I'm using Ubuntu v. 20.04.3 LTS, on Windows 11 using WSL2. Here's my code sample:

`const app = express();

app.get("/connect", (req, res) => {
  const btSerial = new (require('bluetooth-serial-port')).BluetoothSerialPort();
  
  btSerial.on('found', function (address, name) {
    console.log("Found: ");
    console.log(address);
    console.log(name);
  });

  btSerial.on('finished', function () {
      self.emit('error', 'No suitable devices found');
  });

  btSerial.inquire();

  res.json({ message: "Looking for Bluetooth connections" });
});`

To debug, I've imported the segfault-handler package, and it returns the following to my console when I request the /connect endpoint:

`PID 13962 received SIGSEGV for address: 0xb408
/home/valtteri/project/node_modules/segfault-handler/build/Release/segfault-handler.node(+0x3351)[0x7f6f65713351]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x153c0)[0x7f6f681283c0]
node(_ZN2v811HandleScopeC2EPNS_7IsolateE+0x50)[0xceeb00]
/home/valtteri/project/node_modules/bluetooth-obd/node_modules/bluetooth-serial-port/build/Release/BluetoothSerialPort.node(_ZN9DeviceINQ9doInquireEv+0x297)[0x7f6f64705b27]
/home/valtteri/project/node_modules/bluetooth-obd/node_modules/bluetooth-serial-port/build/Release/BluetoothSerialPort.node(_ZN3Nan12AsyncExecuteEP9uv_work_s+0x27)[0x7f6f647071c7]
node[0x1555504]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609)[0x7f6f6811c609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7f6f68043293]
Segmentation fault`

Thanks in advance!

eelcocramer, 2022-07-03:
Can you try moving the `btSerial` const out of the scope of the `get` function? So to the same level of `app`? From the top of my head the module should only be created once throughout the scope of you app. 

Also it may not run well when multiple (web) request must be handled simultaneously. You may want to decouple the bluetooth scanning from the webrequest and return cached results and have a singleton process scan every so much seconds.

valdemartti, 2022-07-03:
Hey. I'll give that a try. Might take the web requests out completely, just to make sure I get the package working at first. Will return to this. Thanks!

valdemartti, 2022-07-05:
Hey. I attempted to comment out all web endpoints, and moved the `btSerial` const to the parent scope, same as `app`. Unfortunately, no luck. The same Segmentation fault persists. Could it be something to do with my computer specs? Perhaps the package does not recognize a Bluetooth driver within my WSL2 instance?

I'll try to run the code directly in my Windows environment next.

closed 2022-07-05 by valdemartti

eelcocramer, 2022-07-05:
Thank you for getting back and update the report.