eelcocramer/node-bluetooth-serial-port #484: Error: The module was compiled against a different Node.js version

ninjadev64, 2023-09-24:
Node versions: 18.18.0, 19.9.0, 20.7.0
Library version: 3.0.2

When I use Node 19.9.0, I get an error saying that the module was compiled against NODE_MODULE_VERSION 111, but my Node version requires 110. 
Similarly, for Node 18.18.0, it's compiled for 108 but states it requires 110. For Node 20.7.0, it's compiled for 115.

108, 110, and 115 as NODE_MODULE_VERSION targets respectively should be correct for those versions - they don't all require 110.

Does anyone know how to resolve this strange behaviour? (@eelcocramer)

ninjadev64, 2023-09-24:
In fact, 18.9.1 is 108, and 19.0.0 is 111. There IS NO 110!

eelcocramer, 2023-09-24:
You must build this module (using `npm install`) with the same version of node that you use to run your code. The binary that is built is not portable between node versions.

ninjadev64, 2023-09-25:
> You must build this module (using `npm install`) with the same version of node that you use to run your code. The binary that is built is not portable between node versions. 

Yes, I am. It always compiles for what should be the right target but then says it requires 110 which doesn't even exist.

ninjadev64, 2023-09-25:
Ah, I've narrowed it down. The library works if I use it on its own, but when I use it with Electron, this error occurs. The latest version of Electron says it requires 116, and the version I'm using (22.3.0) says it requires 110.

I am, however, using other native modules in my project, and they seem to work fine, so it is an issue with this library.

(@eelcocramer)

eelcocramer, 2023-09-25:
Cool that you found a solution. Thanks for the feedback.