eelcocramer/node-bluetooth-serial-port #485: mac  yarn error  help me !!

zhou1591, 2023-09-25:
I wish someone could help me

in macos

run yarn 

active node 14.17.0

error /Users/edy/Desktop/electorn/node_modules/bluetooth-serial-port: Command failed.
Exit code: 1
Command: node-gyp configure build
Arguments: 
Directory: /Users/edy/Desktop/electorn/node_modules/bluetooth-serial-port
Output:
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@14.17.0 | darwin | arm64
gyp info find Python using Python version 3.9.6 found at "/Applications/Xcode.app/Contents/Developer/usr/bin/python3"
gyp info spawn /Applications/Xcode.app/Contents/Developer/usr/bin/python3
gyp info spawn args [
gyp info spawn args   '/Users/edy/.nvm/versions/node/v14.17.0/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/edy/Desktop/electorn/node_modules/bluetooth-serial-port/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/edy/.nvm/versions/node/v14.17.0/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/edy/Library/Caches/node-gyp/14.17.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/edy/Library/Caches/node-gyp/14.17.0',
gyp info spawn args   '-Dnode_gyp_dir=/Users/edy/.nvm/versions/node/v14.17.0/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/Users/edy/Library/Caches/node-gyp/14.17.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/Users/edy/Desktop/electorn/node_modules/bluetooth-serial-port',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o
../src/osx/DeviceINQ.mm:75:16: warning: 'Call' is deprecated [-Wdeprecated-declarations]
        found->Call(2, argv);
               ^
../../nan/nan.h:1747:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:112:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:78:15: warning: 'Call' is deprecated [-Wdeprecated-declarations]
    callback->Call(0, 0);
              ^
../../nan/nan.h:1747:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:112:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:109:16: warning: 'Call' is deprecated [-Wdeprecated-declarations]
    baton->cb->Call(1, argv);
               ^
../../nan/nan.h:1747:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
[... 87 lines cut ...]
../src/osx/DeviceINQ.mm:318:22: error: no matching member function for call to 'Set'
        resultArray->Set(i, deviceObj);
        ~~~~~~~~~~~~~^~~
/Users/edy/Library/Caches/node-gyp/14.17.0/include/node/v8.h:3670:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/edy/Library/Caches/node-gyp/14.17.0/include/node/v8.h:3673:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/osx/DeviceINQ.mm:324:5: warning: ignoring return value of function declared with 'warn_unused_result' attribute [-Wunused-result]
    cb->Call(Nan::GetCurrentContext(), Nan::GetCurrentContext()->Global(), 1, argv);
    ^~~~~~~~ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
7 warnings and 8 errors generated.
make: *** [Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/edy/.nvm/versions/node/v14.17.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:376:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Darwin 21.6.0
gyp ERR! command "/Users/edy/.nvm/versions/node/v14.17.0/bin/node" "/Users/edy/.nvm/versions/node/v14.17.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build"
gyp ERR! cwd /Users/edy/Desktop/electorn/node_modules/bluetooth-serial-port
gyp ERR! node -v v14.17.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok

eelcocramer, 2023-09-25:
Hi. This module requires node v16 or newer. You are trying to build using node v14 and that is not supported anymore. You could try to use an older version if you need to stay on node v14.

closed 2023-09-25 by eelcocramer

zhou1591, 2023-09-26:
> Hi. This module requires node v16 or newer. You are trying to build using node v14 and that is not supported anymore. You could try to use an older version if you need to stay on node v14.

thank you for the response

The version I use is node-bluetooth-serial-port@2.2.7

Because I saw the documentation that macOs needs 2.x installed

Then in the 2.2.7 document, I see that node supports v12-v14 versions

My current node version is v14.17.0, and the command executed is npm i bluetooth-serial-port@2.2.7. My operating system is under the macOs of m1 chip, and then this error occurs. It seems that there is no difference in the document I found. So what am I supposed to do, thank you, big guy, please help me

zhou1591, 2023-09-26:
> Hi. This module requires node v16 or newer. You are trying to build using node v14 and that is not supported anymore. You could try to use an older version if you need to stay on node v14.

Oh, I see another npm dependency, the installation is successful, but he reports another error "not a mach-o file"

Then I noticed that the minimum requirement for m1 is nodeV15 but the library is supported up to V14, is there any way I can use it on m1 with macOs

eelcocramer, 2023-09-26:
Ah, sorry I forgot about the dropped support for MacOS.

I'm afraid it is currently not possible to use the module on recent versions of MacOS. Older MacOS version should still work. Big Sur was probably the latest MacOS version would have worked.

I am not going to try to add support again but am willing to review and accept a pull request that adds support again.

Thanks.