eelcocramer/node-bluetooth-serial-port #486: Error: Error reading from connection

ninjadev64, 2023-09-25:
Oop, I've got another problem.

Every so often, under Ubuntu 22.04.3 LTS, my connection disconnects (and throws the "failure" event) with this error:
`[Error: Error reading from connection]`.

Could you guide me through some simple steps to diagnose the cause of this error please? As you can probably tell, this is my first experience with Bluetooth...

ninjadev64, 2023-09-25:
I'm using the BluetoothSerial Arduino module with an ESP32, if that's relevant information.

eelcocramer, 2023-09-26:
I guess the best way to handle this is to setup a new connection. The reasons that the connection is dropped are well down in the bluetooth stack and radio and are beyond the scope of this module.

ninjadev64, 2023-09-26:
> I guess the best way to handle this is to setup a new connection. The reasons that the connection is dropped are well down in the bluetooth stack and radio and are beyond the scope of this module.

That's what I'm doing, but it's happening abnormally frequently. I know it's not really an issue with this library but I was wondering if you could point me towards some troubleshooting steps :)

eelcocramer, 2023-09-26:
Sorry but I cannot help you with that. It has been quite some years since I worked with bluetooth. Maybe you should [consider asking users on the bluez mailinglist](http://www.bluez.org/contact/) if they know how to troubleshoot? Bluez is the bluetooth stack used on Linux, the module is build on top of that stack.

closed 2023-09-28 by ninjadev64