eelcocramer/node-bluetooth-serial-port #487: Can it possible to send file from one windows machine to another windows machine using node-bluetooth-serial-port

syncthreads-sammith, 2023-12-14:
i have this requirement, where i need transfer file form windows machine to another window
1.I have node v16+
2 .node-gyp v10
3.python v3+
 I have clean setup for this `node-bluetooth-serial-port`  npm package to run and i excuted the code that list out the nearby device this mean that `node-bluetooth-serial-port` is working fine, and i have written code using `node-bluetooth-serial-port` to get external device data here the mode is only reading and successfully got the data working fine  ,but i have another requirement need to send file from my device to another know device **(PC OR TAB windows)** can it possible to send any file (irrespective of file formate)
using **node-bluetooth-serial-port**, i have tried some code **below i've pasted sender code** 
 **Function to search for the receiver device**
```
` function searchForReceiver() {
  console.log('Starting Bluetooth discovery...');
  
  serial.on('found', (address, name) => {
    console.log(`Found Bluetooth device: ${name} (${address})`);

    const { parsedIpAddress } = parseDeviceData(address);

    if (!foundReceiver && parsedIpAddress === receiverAddress) {
      console.log('Receiver device found. Attempting to connect...');
      foundReceiver = true;
        connectToDevice();
    }
  });
  serial.on('finished', () => {
    console.log('Device discovery finished.');
    if (!foundReceiver) {
      console.log('Receiver device not found in the discovered devices.');
      serial.close();
    }
  });

  // Start device discovery
  serial.inquire();
}
`
function connectToDevice() {
  console.log('Connecting to the receiver...');
  serial.findSerialPortChannel(receiverAddress, (channel) => {
    serial.connect(receiverAddress, channel, () => {
      console.log('Connected to Bluetooth device');

      // Replace 'path/to/your/file' with the path to the file you want to send
      const filePath = '/Node-gyp-setUp.txt'; // Replace with the actual file path

      // Read the file and send it over the Bluetooth connection
      const fileStream = require('fs').createReadStream(filePath);
      fileStream.on('data', (data) => {
        serial.write(Buffer.from(data));
      });

      fileStream.on('end', () => {
        console.log('File transfer completed');
        serial.close();
      });
    });
  });
}
searchForReceiver();
```

**Here is receiver code** 
```
function handleIncomingData(buffer) {
  console.log('Received data:', buffer.toString());
  // Process the incoming data as needed
}
// Function to start listening for connections
function startListening() {
  serial.on('data', handleIncomingData);
  serial.on('failure', (error) => {
    console.error(`Error: ${error}`);
  });
  serial.on('close', () => {
    console.log('Connection closed');
    startListening();
  });
}
// Start listening for incoming connections
startListening();
```

I guess sender and receiver handshake isn't happening properly or cloud be some isseue and i don't knon this package is able to do that,so please guid me through,can this be achievable

eelcocramer, 2023-12-14:
Sorry but that is currently not possible. The server is only implemented for LInux.

closed 2023-12-14 by eelcocramer

syncthreads-sammith, 2023-12-15:
Thanks for letting know, do you know, dose node has any npm package that can work for me,Can this sending file possible in nodejs environment ?

eelcocramer, 2023-12-15:
Over bluetooth? Not that I'm aware of no.