eelcocramer/node-bluetooth-serial-port #488: Confusion and Clarification

fromage9747, 2024-03-19:
I am trying to use this module on Ubuntu 22.04 running node v16.

There are quite a few old issues stating that there is no support past 13, but it installed without issue on 16, no errors. Then I see there have been updates since the previous mentions of issues past 13 and in the description stating that there is no support past 13. 

Can I get a final confirmation in 2024 that this repo requires node13 or lower in order to function? Or the code runs, just that I cannot discover my device when running the server.  Then again, maybe I am running it wrong.

I am using NestJS and I activate the Bluetooth code via a route, however the server object is pretty empty once it's called.  

```
{
        "_events": {},
        "_eventsCount": 1,
        "inDisconnect": false,
        "server": {}
    }
```

I have gone through the other issues that have been reported here and nothing has changed the outcome.

fromage9747, 2024-03-19:
Thank you for the clarification regarding node version. I guess I need to wait for the other chap that built the server side of things. @atilag

fromage9747, 2024-03-19:
@eelcocramer I forgot to mention that I tried with the Client as well and it was pretty much the same thing.

This is the result of the client:

```
{
        "_events": {},
        "_eventsCount": 1,
        "inq": {}
    }
```

eelcocramer, 2024-03-19:
The result of what call?

fromage9747, 2024-03-19:
If I console log btSerial from "var btSerial = new (require("bluetooth-serial-port").BluetoothSerialPort)();" 

Just there is no output that any devices have been discovered etc. I do have devices available.  If I use the built-in OS Bluetooth menu I discover devices.  If I use bluetoothctl I discover devices.

I don't get any output really when I run the example code or modified code that I have found from other users.

eelcocramer, 2024-03-19:
I believe it only discovers devices that have the serial port profile.